CJNE/ha-sunspec #14: scaling factor issue in DC power of strings

der-optimist, 2021-06-25:
I have a Fronius Gen 24 inverter (10 kW) with 2 MPPTs (strings). The scaling factor of the DC power of these strings seems not to be correct:
[image]
[image]
above 600 W it's hopping to 60 W

CJNE, 2021-06-26:
Thanks for your report! 
This is most likely a bug in the inverter itself or the upstream library (pysunspec2) as the component itself does not directly deal with scaling factor calculations. 
So to begin with i think we should make them (SunSpec) aware of the issue and then try to figure out wether it makes sense to make a special case workaround for this particular inverter.
We should try to isolate and reproduce the problem, would you be able to run a python script that a write during the time your dc power around 600W in order to get some data for a bug report?

der-optimist, 2021-06-28:
My current config is using Modbus with manually configured sensors:
https://github.com/der-optimist/hass/blob/master/modbus.yaml
From that, I calculate the String DC power:
`
"{{ ((states('sensor.fronius1_modbus_power_dc_string1_raw') | int) * (10 ** ((states('sensor.fronius1_modbus_power_dc_scaling_factor')| int )) )) |round(0)}}"
`
Here I don't see that problem. So I don't think this is an inverter bug. 

Python script: Do you have a script I could use? What do I need for that?

mptei, 2021-07-16:
I have a Fronius Symo with a similar problem. It shows 20.000W instead of 2000W (for instance). In this situation I created a Jupyter notebook to get the values directly with the pysunspec2 lib. There, printing cvalue showed the correct results.

mptei, 2021-07-16:
For instance: HA shows 583W, but
```
import sunspec2.modbus.client as client
d = client.SunSpecModbusClientDeviceTCP(slave_id=1, ipaddr='wreastwest', ipport=502)
d.scan()
d.common[0].read()
print(d.mppt[0].module[1].DCW.cvalue)
d.close()
```

prints 58.300000000000004

mptei, 2021-07-16:
I debugged into the component and saw that here even cvalue has the wrong value. The raw value and the scale factor are matching to the cvalue. So I guess the problem is that the scale factor, which is part of another model (?), is not in sync with the mppt model.

mptei, 2021-07-16:
After letting the Jupyter notebook running for a longer time (to get a scale factor change) I see, that also here the wrong value is delivered. So for sure the problem is in the pysunspec2 lib.

mptei, 2021-07-17:
I've created an issue in the sunspec2 lib: [https://github.com/sunspec/pysunspec2/issues/42](url)

CJNE, 2021-07-17:
That’s great, thank you for all the help! Hopefully they will be able to
fix this soon, otherwise maybe I can figure out a workaround in the
meantime.

thecem, 2021-08-15:
Same Issue here for current...

If you need more info let me know!

Thanks a lot!

mptei, 2021-08-15:
@thecem With the PR [https://github.com/sunspec/pysunspec2/pull/43](https://github.com/sunspec/pysunspec2/pull/43) it works for me. Please comment there as it is a sunspec library problem.

thecem, 2021-08-15:
Ok, I updated to 0.0.8 (same problem as before HACS shows updated for 0.0.8 to 0.0.7 is available!) and the current values are correct now....
But the versioning have the same problem, this was also the reason I missed the update to 0.0.8.....
What is the problem here with the versioning?

Thanks a lot!

thecem, 2021-08-15:
[image]

mptei, 2021-08-15:
> and the current values are correct now...
Yes, because with the restart the scale factor is updated. But when your sunspec device changes the sf, the values will be wrong again.

thecem, 2021-08-15:
Ok, so will be there a solution for dynamically update the sf if the sunspec device will change it?

with pysunspec2==1.0.5 it is solved?

mptei, 2021-08-15:
If pysunspec2 1.0.5 would fix [https://github.com/sunspec/pysunspec2/issues/42](https://github.com/sunspec/pysunspec2/issues/42) than yes.

You might checkout the PR [https://github.com/sunspec/pysunspec2/pull/43](https://github.com/sunspec/pysunspec2/pull/43) and comment there.

thecem, 2021-08-15:
I forked sunspec/pysunspec2, changed the device.py and released to 1.0.5

and changed in manifest.json:

  "requirements": ["thecem/pysunspec2==1.0.5"]

But not working....

thecem, 2021-08-15:
with manifest.json: 

`"requirements": ["git+https://github.com/thecem/pysunspec2@master#pysunspec2==1.0.5"]`

like described in: https://developers.home-assistant.io/docs/creating_integration_manifest/#custom-requirements-during-development--testing

it is working , since my sensor are available again.

Additional it tested with 1.0.4 (without PR sunspec/pysunspec2#43 ) and with your proposed changes.
With your changes the sf will be updated and the values will be correct calculated.
I will monitor this for 24 h.

Hopefully your PR will be merged into master and tagged to 1.0.5 !

Thanks a lot!

mptei, 2021-08-15:
Please comment on [https://github.com/sunspec/pysunspec2/pull/43](https://github.com/sunspec/pysunspec2/pull/43). This would help to get this PR accepted.

CJNE, 2021-08-15:
> Ok, I updated to 0.0.8 (same problem as before HACS shows updated for 0.0.8 to 0.0.7 is available!) and the current values are correct now....
> But the versioning have the same problem, this was also [...]

Thanks for pointing that out, it seems to happen when i create a new release using the draft created by the github action instead of creating it from scratch, not really sure what's up with that. I have created a new 0.0.8 release now and it seems to work.

CJNE, 2021-08-15:
> Please comment on [sunspec/pysunspec2#43](https://github.com/sunspec/pysunspec2/pull/43). This would help to get this PR accepted.

Great job, i hope they will accept your PR soon!

Kunvert, 2021-09-15:
I have the same issue with a Solaredge inverter. I would like to try to modify the code by miself, but i cannot find the file device.py in the folder /custom_component/sunspec. Moreover i cannot even find the update 1.0.5 for pysunspec2. Can somebody please help. Many Thanks!

thecem, 2021-09-15:
You can use my repo where the fix is implemented, only change in

/custom_component/sunspec/manifest.json:

`"requirements": ["git+https://github.com/thecem/pysunspec2@master#pysunspec2==1.0.5"]`

And you are done…

Kunvert, 2021-09-15:
Great! Thank you. I will check what happens in the next hours

thecem, 2021-09-15:
If positiv, pls address this to the main repo under:

Please comment on sunspec/pysunspec2#43. This would help to get this PR accepted.

purcell-lab, 2021-10-07:
Same issue here, with SolarEdge 5 kW inverter. I'll apply the workaround and report back.

Interestingly the SolarEdge MODBUS HA integration doesn't have thus issue.

[image]

thecem, 2021-12-10:
[sunspec/pysunspec2#42](https://github.com/sunspec/pysunspec2/issues/42) is closed.

try to use 1.0.6, update here if it will work!

`"requirements": ["pysunspec2==1.0.6"]`

closed 2022-02-07 by CJNE