CJNE/ha-sunspec #16: Interval 30sec

thecem, 2021-07-10:
**Is your feature request related to a problem? Please describe.**
 Update Interval should be configure able

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Maybe a setting per device or a central setting for:
update-interval: 10 sec

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Is there an option? What is the interval?

thanks a lot!

CJNE, 2021-07-15:
Thanks for the suggestion! It has been implemented in version 0.0.5, published just now.

closed 2021-07-15 by CJNE

thecem, 2021-07-16:
Thanks a lot!
Interval of “0” is default and so 30 sec?

CJNE, 2021-07-16:
Yes 30 is the default, it's probably showing 0 for you since you already had it set up, a little bug that should not affect functionality :)

commit 52a303a1 (2021-07-15): Configurable scan interval, resolve #16