CJNE/ha-sunspec #17: Fronius meter model (240) not showing up in selection

Chipa001, 2021-07-15:
I have a Fronius inverter & meter installed.
I can manually query both model 1 (Inverter) and 240 (meter) without issue.

The Integration is working well for the Inverter, but model 240 (meter) does not show up in the selection.

I'm not proficient at coding so I haven't figured out how to trace what may be missing.

CJNE, 2021-07-15:
What models are showing up in the select menu? (either in step 2 of initial setup or if you click "Configure" from the integrations screen)
Model 240 is not enabled by default but if your inverter supports it it should be able to add it using that select menu, in theory at least :)

Chipa001, 2021-07-15:
1,111,120,121,122,123,160 all turn up in the selection box. Only 1,111 & 160 having useful live readings but they all work otherwise.
Using QModMaster I'm scanning slave 240 address 40071->40192 without an issue. I've got a NodeRed process reading data out atm, but it is a pain to maintain in its current form. I've already used this to replace the nodered readings for the inverter, so just the Meter is left now.

CJNE, 2021-07-15:
It sounds like the inverter and meter are actually two separate modbus slaves, the inverter having slave id 1 and the meter have slave id 240?
In that case it should be possible to have to set up two devices, first add the inverter using slave id 1 and then create another for the meter using slave id 240. The meter, if it supports SunSpec, should have models in the 200 range (most likely model 201 for a single phase meter).
Does that help?

thecem, 2021-07-16:
Yes that right the ID of the first fronius meter is on a Symo Hybrid 240, not the model. The first inverter is ID 1.
If you have a gen24 the meter starts with ID 200.

Chipa001, 2021-07-16:
Both slaves are addressed by the one interface card. e.g. x.x.x.x:502
The integration scans for slaves, currently finds 1,111,120,121,122,123,160, but not 240.
How do I setup a second device to the same card and get a different result?

> It sounds like the inverter and meter are actually two separate modbus slaves, the inverter having slave id 1 and the meter have slave id 240?
> In that case it should be possible to have to set up [...]

closed 2021-07-16 by Chipa001

CJNE, 2021-07-16:
As you figured out it does not scan for slaves, it scans for SunSpec models (Modbus registers).
I haven't been able to test multiple instances of the integration so it's nice to hear that it seems to work!

Chipa001, 2021-07-16:
Give me 24 hours to confirm. Bad weather right now, so only the meter is giving me real readings. Once I get some sunlight I'll be able to confirm both are working simultaneously as expected.

thecem, 2021-07-16:
You have to setup two integrations one with ID 1 and one with ID 240, regardless the weather you have instantly the values…

thecem, 2021-07-16:
[image]

may it help if the ID is presented in the overview behind the up-port.

Chipa001, 2021-07-17:
> You have to setup two integrations one with ID 1 and one with ID 240, regardless the weather you have instantly the values…

Correct, but until both are showing live data changing frequently (e.g good solar weather) there is no proof both are working correctly with good refresh rates.

Chipa001, 2021-07-17:
> [image]
> 
> 
> 
> may it help if the ID is presented in the overview behind the ip-port.
> 
> 

Agreed. Multiple ID's behind the same interface is messy.