CJNE/ha-sunspec #18: SMA Inverter - Modbus exception 2: addr: 40085 count: 98

Tonno87, 2021-07-18:
## Version of the custom_component

0.0.5

## Configuration

```yaml

SMA inverter Tripower 8.0
ip 192.168.178.85
port 502
Sunspec ID 126 (from manual)

```

## Describe the bug

ha-sunspec cannot connect to my SMA inverter. The sunspec support starts with registers above 40001, so the ID address is 40086, no 40085 exists. I think this is the main cause.

On the manual they said:
```
Das SunSpec Modbus-Profil für SMA
Produkte beginnt ab der Registernummer 40001.
```

## Debug log

```text

Logger: custom_components.sunspec
Source: custom_components/sunspec/config_flow.py:121
Integration: SunSpec (documentation, issues)
First occurred: 15:47:57 (3 occurrences)
Last logged: 15:52:35

Failed to connect to host 192.168.178.85:502 - Modbus exception 2: addr: 40085 count: 98

```

CJNE, 2021-07-18:
This looks like a problem with [pySunSpec2](https://github.com/sunspec/pysunspec2) (the library that this integration relies on for the modbus communication), can you create an issue over there?

Tonno87, 2021-07-18:
There's already an issue on pySunSpec2:
[Modbus exception with d.scan() on SMA inverter #36](https://github.com/sunspec/pysunspec2/issues/36#issue-901868774)

But it seems to work with an [offset](https://github.com/sunspec/pysunspec2/issues/36#issuecomment-849674716).