CJNE/ha-sunspec #22: Not able to connect to Solaredge SE10K inverter

marschallphilipp, 2021-07-26:
I´m not able to connect to my SE10K inverter from Solaredge although it implements the sunspec implementation.
I disabled and reenabled the Modbus TCP connection and tried to connect through the integration within a timespan of 2minutes but can´t connect to the inverter.
[image]

## Version of the custom_component
0.0.5 (latest)

## Configuration
IP Address from inverter (192.168.2.53)
Port (1502 was set in Solaredge SetApp)

CJNE, 2021-07-28:
Have you checked that the slave id matches your inverter device id?
Can you ping it on that IP?

marschallphilipp, 2021-07-29:
What do you mean by"inverter device id"? I think I can remember that solar edge inverts just accepts one modbus slave at a time and this should be always "1". Am I wrong?
I didn't try to ping it but I know that the IP is correct since I successfully managed to get this custom component working:
https://github.com/erikarenhill/solaredge-modbus-hass

marpi82, 2021-08-01:
I have the same issue with SolarEdge SE7K, inverter has working on solaredge-modbus-hass integration but on Your integration it won't work
[image]
[image]

CJNE, 2021-08-01:
> What do you mean by"inverter device id"? I think I can remember that solar edge inverts just accepts one modbus slave at a time and this should be always "1". Am I wrong?
>
> I didn't try to ping it [...]

I meant modbus slave id, the other component is using 1 so that shouldn't be the problem.
The client library used in my component, pySunSpec2, is doing a scan on initial connection to discover what registers are available. A theory is that perhaps this takes to long or trips up your inverter somehow. 
It would be interesting to see if a python script using pySunSpec2 can connect to it outside of Home Assistant. I'm currently not home but once I get back I'm happy to try and works something out to test this.

CJNE, 2021-08-01:
> I have the same issue with SolarEdge SE7K, inverter has working on solaredge-modbus-hass integration but on Your integration it won't work
> 
> [image]
> 
> [image]
> 
> 
Sounds like the same problem as above, would be happy to try and figure this out once I get home again.

dhover, 2021-08-28:
I'am using 0.0.11 and changed the two lines in api.py:

FROM:
```
model.read() 
time.sleep(0.6) 
```

TO:
```
time.sleep(0.6) 
model.read() 
```

Now the integration is working for my SolarEdge SE5K SetApp inverter.

marpi82, 2021-08-28:
> I'am using 0.0.11 and changed the two lines in api.py:
> 
> FROM:
> 
> ```
> model.read() 
> time.sleep(0.6) 
> ```
> 
> TO:
> 
> ```
> time.sleep(0.6) 
> model.read() 
> ```
> 
> Now the integration is working for my SolarEdge SE5K SetApp inverter.

On 0.0.11 these two lines are:
```
time.sleep(0.6) 
model.read() 
```
not need to change to... You really have a 0.0.11 with
```
model.read() 
time.sleep(0.6) 
```
?

https://github.com/CJNE/ha-sunspec/blob/80a468f4eb2549f131e7a540f23652dc07b1b778/custom_components/sunspec/api.py#L130-L132

dhover, 2021-08-28:
You're right. I don't know what happened. But it is working now with the SolarEdge inverter and in the previous versions it wasn't.

marschallphilipp, 2021-09-02:
> Now, version 0.0.11 installed from HACS is working for me without aby problems and modifications :). How about You @marschallphilipp ?

Thank you all for making this possible. I can confirm that it's more working for me with a SE10K inverter from Solaredge.
To be honest, I hoped to get some data of the BYD LVS batteries which are corrected to the converter too. Unfortunately this is not the case.
When trying to receive data from the batteries using the modbus integration I haven't been lucky. Unfortunately I haven't found any reporting that this is actually possible with the SE10K inverter but still hoping to get it working some day soon.
Anyway thank you very much!