CJNE/ha-sunspec #32: Unable to prepare setup for platform sunspec.sensor

dhover, 2021-08-12:
## Version of the custom_component:
0.0.7

## Describe the bug

After upgrading from 0.0.5 to 0.0.7 an error is thrown.

## Config
HA core-2021.7.4
HA OS 6.1
Fronius Galvo 1.5-1 inverter

## Debug log

Logger: homeassistant.setup
Source: setup.py:311
First occurred: 4:46:09 PM (1 occurrences)
Last logged: 4:46:09 PM

Unable to prepare setup for platform sunspec.sensor: Platform not found (cannot import name 'ELECTRIC_CURRENT_AMPERE' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)).

TravisWilder, 2021-08-13:
Hi 
I think the ELECTRIC_CURRENT_AMPERE is part of the 2021.8 Core and you should consider to upgrade
or not go for the latest version of the integration

CJNE, 2021-08-13:
Yes, as @TravisWilder pointed out, from version 0.0.7 you need to be running HA 2021.8, they introduced some breaking changes in that one. 0.0.6 should be just as good if you don't want to update HA yet.

closed 2021-08-13 by CJNE

dhover, 2021-08-14:
I've updated to 2021.8. I can confirm it's working now!