CJNE/ha-sunspec #35: Integration not found

MrJoki007, 2021-08-16:
Version: 0.0.8
I installed this integration with Homeassistant HACS.

But when i try to search for it in the integrations tab i dont find anything.

Input of log:
``2021-08-16 09:20:06 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration sunspec which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant``

CJNE, 2021-08-16:
That sounds strange, did you restart Home Assistant after installing it?
Version 0.0.8 requires HA 2021.8 or newer, are you on that version?

MrJoki007, 2021-08-16:
Using version 2021.8.6.
Yes and i did restart homeassistant.
I tried with another integration of HACS and that shows up as normal

CJNE, 2021-08-16:
And you search for sunspec right? 
If it don't show up then something might prevent it from loading, are there any other messages related to this integration in the log?

MrJoki007, 2021-08-16:
I got a notification right now, that version 2021.8.7 is available right now.
I did a upgrade and it shows up now.
Thanks

closed 2021-08-16 by MrJoki007