CJNE/ha-sunspec #40: Wrong values with Fronius inverter

dhover, 2021-08-28:
## Version of the custom_component

0.0.9

## Configuration

## Describe the bug

The value for AC power for the Fronius inverter are sometimes too high. More than the capacity of the unit.
My panels have a capacity of 1530 Wp.  The values goes as high as 3000 Watt!
It looks the values looks 10 times too high. Also the DC Power values have those spikes.
The DC Voltage and DC Current values seems ok.
What can be wrong?

AC Power
[image]

AC Current
[image]

DC Power
[image]

DC Voltage
[image]

DC Current
[image]

## Debug log

dhover, 2021-08-28:
Extra info: I'm using a Fronius Galvo 1.5-1 inverter.

dhover, 2021-08-28:
I now see there is a similiar issue #14  

`with manifest.json:
"requirements": ["git+https://github.com/thecem/pysunspec2@master#pysunspec2==1.0.5"]`

I've made the suggested change and will examine the values after a day to see if the spikes are gone.

dhover, 2021-09-16:
The spikes are gone. So the modified pysunspec2 library seems to solve the problem.

CJNE, 2021-11-19:
Great, let's hope they merge the fix upstream as well soon!

closed 2021-11-19 by CJNE