CJNE/ha-sunspec #44: Energy units

marpi82, 2021-09-03:
Hello, could you force the energy in watts per hour to be forced to display in kilowatts per hour? 
Only kWh is correctly calculated by the energy module in the Home Assistant
[image]

Spitfire69, 2021-09-03:
Just change it in your customize.yaml file:
sensor.inverter_ac_energy:
  unit_of_measurement: 'Wh'

marpi82, 2021-09-03:
> Just change it in your customize.yaml file:
> sensor.inverter_ac_energy:
> unit_of_measurement: 'Wh'

The value and the unit is displayed correctly, until today I have produced 12.9MWh, i.e. 1294kWh, i.e. 12941668Wh. So everything is correct. But it should be displayed in HA in kWh units only (not in Wh) that the energy module could correctly interprets the value of energy produced.

If the value is taken from inverter in Wh, it should be divided by 1000 and displayed in kWh. These is the problem

marpi82, 2021-09-03:
[...] As You can see here.

Solar production energy **(kWh)**
[image]

CJNE, 2021-09-03:
My inverter also has solar energy in Wh and that works fine with the energy dashboard, can you look at the entity you expected to use there in the developer tools and look at what attributes it has?

Spitfire69, 2021-09-03:
Yes, it works with both Wh and kWh, if you have problem it’s likely something else however I had the same problem as you until I changed like I wrote above. Have you tried it?

CJNE, 2021-09-03:
I'm not agains changing to kWh, it's just that it might not be the issue. Ideally you shouldn't have to do the customization step so perhaps there is something that needs to be changed.

marpi82, 2021-09-04:
> Yes, it works with both Wh and kWh, if you have problem it’s likely something else however I had the same problem as you until I changed like I wrote above. Have you tried it?

@Spitfire69 in fact, you are right.  I have suggested with this information in parentheses that energy should be in kWh without checking this.  Even when it is - as in this case - in Wh, everything is converted correctly by HA.

So I'am closing the issue - because I also do not care about displaying this value in a specific unit. I didn't have to customize anything and everything is fine.

I just made a wrong suggestion on what was written without checking this. Sorry

closed 2021-09-04 by marpi82

CJNE, 2021-09-05:
No worries :)