CJNE/ha-sunspec #47: Unexpected error fetching sunspec data

dhover, 2021-09-26:
## Version of the custom_component

0.0.13

## Configuration

## Describe the bug

Most of the days I got one exception for the Sunspec integration.
I have installed the SunSpec integrartion two times, one for the Fronius inverter and one for the solarEdge inverter.
I can't see in the logfile which one is producing this error.

## Debug log

`2021-09-25 21:20:25 WARNING (MainThread) [custom_components.sunspec] Response timeout
2021-09-25 21:20:25 ERROR (MainThread) [custom_components.sunspec] Unexpected error fetching sunspec data: Connection error: [Errno 111] Connection refused
Traceback (most recent call last):
File "/config/custom_components/sunspec/__init__.py", line 128, in _async_update_data
data[model_id] = await self.api.async_get_data(model_id)
File "/config/custom_components/sunspec/api.py", line 86, in async_get_data
return await self.read(model_id)
File "/config/custom_components/sunspec/api.py", line 89, in read
return await self._hass.async_add_executor_job(self.read_model, model_id)
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/sunspec/api.py", line 132, in read_model
model.read()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 85, in read
data = self.model.device.read(self.model.model_addr + self.offset, self.len)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 317, in read
return self.client.read(addr, count, op)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 584, in read
data = self._read(addr + read_offset, read_count, op=op)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 529, in _read
raise ModbusClientError('Response timeout')
sunspec2.modbus.modbus.ModbusClientError: Response timeout
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 481, in connect
self.socket.connect((self.ipaddr, self.ipport))
ConnectionRefusedError: [Errno 111] Connection refused
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 187, in _async_refresh
self.data = await self._async_update_data()
File "/config/custom_components/sunspec/__init__.py", line 133, in _async_update_data
self.api.reconnect()
File "/config/custom_components/sunspec/api.py", line 106, in reconnect
client.connect()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 308, in connect
self.client.connect()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 483, in connect
raise ModbusClientError('Connection error: %s' % str(e))
sunspec2.modbus.modbus.ModbusClientError: Connection error: [Errno 111] Connection refused`

Spitfire69, 2021-09-26:
Seen the same error myself (Kaco inverter).

closed 2021-09-27 by CJNE

dhover, 2021-09-28:
I installed the new version. Thanks for that.

commit c3d9671e (2021-09-27): Don't  log full exception on connection timeout, fixes #47