CJNE/ha-sunspec #48: Connection refused

dhover, 2021-09-28:
## Version of the custom_component

0.0.14

## Configuration

## Describe the bug

An exception error "Connection refused" shows up. I think it is triggered as soon as the inverter starts in the morning.
I've also seen those errors on the end of the day when the inverters stops.

## Debug log

`2021-09-28 07:25:36 WARNING (MainThread) [custom_components.sunspec] Response timeout
2021-09-28 07:25:36 ERROR (MainThread) [custom_components.sunspec] Unexpected error fetching sunspec data: Connection error: [Errno 111] Connection refused
Traceback (most recent call last):
File "/config/custom_components/sunspec/__init__.py", line 129, in _async_update_data
data[model_id] = await self.api.async_get_data(model_id)
File "/config/custom_components/sunspec/api.py", line 97, in async_get_data
return await self.read(model_id)
File "/config/custom_components/sunspec/api.py", line 104, in read
return await self._hass.async_add_executor_job(self.read_model, model_id)
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/sunspec/api.py", line 147, in read_model
model.read()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 85, in read
data = self.model.device.read(self.model.model_addr + self.offset, self.len)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 317, in read
return self.client.read(addr, count, op)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 584, in read
data = self._read(addr + read_offset, read_count, op=op)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 529, in _read
raise ModbusClientError('Response timeout')
sunspec2.modbus.modbus.ModbusClientError: Response timeout
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 481, in connect
self.socket.connect((self.ipaddr, self.ipport))
ConnectionRefusedError: [Errno 111] Connection refused
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 187, in _async_refresh
self.data = await self._async_update_data()
File "/config/custom_components/sunspec/__init__.py", line 138, in _async_update_data
self.api.reconnect()
File "/config/custom_components/sunspec/api.py", line 121, in reconnect
client.connect()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 308, in connect
self.client.connect()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 483, in connect
raise ModbusClientError('Connection error: %s' % str(e))
sunspec2.modbus.modbus.ModbusClientError: Connection error: [Errno 111] Connection refused
2021-09-28 07:26:06 WARNING (MainThread) [custom_components.sunspec] Socket write error: [Errno 32] Broken pipe`

CJNE, 2021-09-30:
I have release 0.0.15 as a pre-release (can be installed in HACS bu choosing Reinstall, show beta versions).
The connection handling has been tweaked and connection errors should look a little nicer, but would be good with some testing before official release.

dhover, 2021-10-01:
I've installed 0.0.15

Error yesterday evening:

`2021-09-30 21:04:55 WARNING (MainThread) [custom_components.sunspec] Response timeout
2021-09-30 21:04:58 ERROR (MainThread) [custom_components.sunspec] Unexpected error fetching sunspec data: Connection error: [Errno 111] Connection refused
Traceback (most recent call last):
File "/config/custom_components/sunspec/__init__.py", line 129, in _async_update_data
data[model_id] = await self.api.async_get_data(model_id)
File "/config/custom_components/sunspec/api.py", line 97, in async_get_data
return await self.read(model_id)
File "/config/custom_components/sunspec/api.py", line 104, in read
return await self._hass.async_add_executor_job(self.read_model, model_id)
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/sunspec/api.py", line 147, in read_model
model.read()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 85, in read
data = self.model.device.read(self.model.model_addr + self.offset, self.len)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 317, in read
return self.client.read(addr, count, op)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 584, in read
data = self._read(addr + read_offset, read_count, op=op)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 529, in _read
raise ModbusClientError('Response timeout')
sunspec2.modbus.modbus.ModbusClientError: Response timeout
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 481, in connect
self.socket.connect((self.ipaddr, self.ipport))
ConnectionRefusedError: [Errno 111] Connection refused
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 187, in _async_refresh
self.data = await self._async_update_data()
File "/config/custom_components/sunspec/__init__.py", line 138, in _async_update_data
self.api.reconnect()
File "/config/custom_components/sunspec/api.py", line 121, in reconnect
client.connect()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 308, in connect
self.client.connect()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 483, in connect
raise ModbusClientError('Connection error: %s' % str(e))
sunspec2.modbus.modbus.ModbusClientError: Connection error: [Errno 111] Connection refused
2021-09-30 21:05:28 WARNING (MainThread) [custom_components.sunspec] Socket write error: [Errno 32] Broken pipe`

Error this morning:

`2021-10-01 07:05:28 WARNING (MainThread) [custom_components.sunspec] Response timeout
2021-10-01 07:05:28 ERROR (MainThread) [custom_components.sunspec] Unexpected error fetching sunspec data: Connection error: [Errno 111] Connection refused
Traceback (most recent call last):
File "/config/custom_components/sunspec/__init__.py", line 129, in _async_update_data
data[model_id] = await self.api.async_get_data(model_id)
File "/config/custom_components/sunspec/api.py", line 97, in async_get_data
return await self.read(model_id)
File "/config/custom_components/sunspec/api.py", line 104, in read
return await self._hass.async_add_executor_job(self.read_model, model_id)
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/sunspec/api.py", line 147, in read_model
model.read()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 85, in read
data = self.model.device.read(self.model.model_addr + self.offset, self.len)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 317, in read
return self.client.read(addr, count, op)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 584, in read
data = self._read(addr + read_offset, read_count, op=op)
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 529, in _read
raise ModbusClientError('Response timeout')
sunspec2.modbus.modbus.ModbusClientError: Response timeout
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 481, in connect
self.socket.connect((self.ipaddr, self.ipport))
ConnectionRefusedError: [Errno 111] Connection refused
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 187, in _async_refresh
self.data = await self._async_update_data()
File "/config/custom_components/sunspec/__init__.py", line 138, in _async_update_data
self.api.reconnect()
File "/config/custom_components/sunspec/api.py", line 121, in reconnect
client.connect()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/client.py", line 308, in connect
self.client.connect()
File "/usr/local/lib/python3.9/site-packages/sunspec2/modbus/modbus.py", line 483, in connect
raise ModbusClientError('Connection error: %s' % str(e))
sunspec2.modbus.modbus.ModbusClientError: Connection error: [Errno 111] Connection refused
2021-10-01 07:05:58 WARNING (MainThread) [custom_components.sunspec] Socket write error: [Errno 32] Broken pipe`

CJNE, 2021-10-01:
Something went wrong when i published 0.0.15, for some reason 0.0.14 was used..
I've published 0.0.16 as a pre-release now, it seems to work better this time :)

dhover, 2021-10-01:
I've installed 0.0.16 and I'll wait till tomorrow!

CJNE, 2021-10-01:
Great, thanks for your help!

dhover, 2021-10-02:
The logs from last night:

`2021-10-01 20:54:52 WARNING (MainThread) [custom_components.sunspec] [Errno 104] Connection reset by peer
2021-10-01 20:54:52 ERROR (MainThread) [custom_components.sunspec] Error fetching sunspec data:
2021-10-01 20:55:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 20:55:52 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 20:56:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 20:56:52 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 20:57:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 20:57:52 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 20:58:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 20:58:52 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 20:59:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 20:59:52 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 21:00:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 21:00:52 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 21:01:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-01 21:01:52 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error`

So every 30 sec a warning till now

`2021-10-02 10:10:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-02 10:10:52 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-02 10:11:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-02 10:11:52 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-02 10:12:22 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error`

I don't know why there are warnings at this moment because the inverter is running normally. It looks the SunSpec integration is not. All sensors are unavailable for the SolarEdge and for the Fronius inverter a lot of them are also unavailable.

After a restart of HA everything looks normal again. So no warnings and all sensors are running!

dhover, 2021-10-03:
Last night I've got the same problems again.

`2021-10-02 20:54:01 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-02 20:54:01 ERROR (MainThread) [custom_components.sunspec] Error fetching sunspec data:
2021-10-02 20:54:31 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-02 20:55:01 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-02 20:55:31 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-02 20:56:01 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-02 20:56:31 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error`
...
`2021-10-03 09:54:31 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-03 09:55:01 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error
2021-10-03 09:55:31 WARNING (MainThread) [custom_components.sunspec] SunSpec modbus connect error`

CJNE, 2021-10-04:
Not quite sure how to address this, if your inverter shuts down during night then it's natural that it can't connect.
But once it's up and running again the sensors should becoma available again, so there might be a problem still.
At that stage the sensors are unavailable but there are no more warnings?

So should it be logged or just completely ignored? If we just ignore then it might make it difficult to trouble shoot "real" connection issues.

One approach might be to disable polling and use an automation to schedule updates instead, described here:
https://www.home-assistant.io/blog/2021/06/02/release-20216/#disable-polling-updates-on-any-integration
The idea is that you can then skip updating during night time for example.

dhover, 2021-10-05:
I've returned to version 0.0.14. With that version I get one exception in the evening and one exception in the morning. During the night no entries in the logfile for this integration. If I change to 0.0.16 I get an entry in the logfile every time the integration polls. So version 0.0.14 is almost what is best. Only the two exceptions should be handled. Then it is perfect.
My two inverters are not turned off during the night. One remark: because I have 2 inverters (SolarEdge and Fronius) I can't see which inverter is producing the entries in the logfile. Is it possible to add some extra info (ip address?) to the logfile so you can see which inverter is producing the warnings/errrors.

[image]

In the morning the sensors didn't become available with 0.0.16. There are no more warnings. I've send all of them.
With 0.0.14 the sensors do become available. 

Disable polling completely is not a good solution in my opinion because then you have to poll every cycle with the help of an automation. If there was a HA service to enable and disable polling then you could disable polling in the evening and enable in the morning. That would be a good solution for more integrations!