CJNE/ha-sunspec #62: Installation with E3DC S10 (SunSpec mode) does not work

SaschaHenning, 2021-11-13:
## Version of the custom_component
`0.0.18`

## Configuration

Not available since installation of intregration does not work.

## Describe the bug

System:
- E3DC S10 ( https://www.e3dc.com )
- Modbus Mode: Sunspec ( http://sunspec.org/wp-content/uploads/2016/02/SunSpec-Information-Model-Reference.xlsx )
- Configuration: https://www.loxwiki.eu/download/attachments/9339566/ModuBus-Dokumentation_2016-08-05.pdf?version=1&modificationDate=1471507657000&api=v2 -> see chapter 3.2 for sunspec config

Error occurs when trying to add the integration on "/config/integrations". It cannot install due to `Failed to connect to host 192.168.0.245:502 slave 1 - Repeating group count not consistent with model length for model 803,model repeating len = 6, model repeating group len = 32`(see logs below)

*Did anyone successfully use this integration for an E3DC device?*

It works fine when I use the modbus integration for direct register access. Here a part of my configuration (for details compare with excel above):

```yaml
modbus:
  - name: E3DC
    type: tcp
    host: 192.168.0.245
    port: 502
    sensors:
      - name: e3dc_meter_home_incoming
        unit_of_measurement: kWh
        address: 40248 #40203+46-1
        slave: 1
        device_class: energy
        count: 2
        scan_interval: 15
        data_type: uint32
        scale: 0.001
        precision: 2
      - name: e3dc_meter_home_exporting
        unit_of_measurement: kWh
        address: 40240 #40203+38-1
        slave: 1
        device_class: energy
        count: 2
        scan_interval: 15
        data_type: uint32
        scale: 0.001
        precision: 2
...
```

## Debug log

```text
2021-11-13 21:03:09 DEBUG (MainThread) [custom_components.sunspec] New SunspecApi Client
2021-11-13 21:03:09 DEBUG (SyncWorker_1) [custom_components.sunspec] Client connect to IP 192.168.0.245 port 502 slave id 0 using timeout 60
2021-11-13 21:03:09 DEBUG (SyncWorker_1) [custom_components.sunspec] Client connected, perform initial scan
2021-11-13 21:03:09 ERROR (MainThread) [custom_components.sunspec] Failed to connect to host 192.168.0.245:502 slave 1 - Repeating group count not consistent with model length for model 803,model repeating len = 6, model repeating group len = 32
```

CJNE, 2021-12-15:
Hi @SaschaHenning 
There is a good chance that this problem has been fixed by the newly released pySunSpec2 release 1.0.6.
The latest release of this integration uses that version so you might have better luck using that?

SaschaHenning, 2022-02-01:
No changes -> tried again today

CJNE, 2022-02-07:
OK, then the best chance to get this resolved is to file an issue at pySunSpec 2, the library we rely on for this.
https://github.com/sunspec/pysunspec2/issues
Include that error message about repeating group, it might be a bug in the SunSpec model, or in the inverter firmware.

CJNE, 2022-02-07:
Try the latest release, it's using an updated version of pySunSpec2 that might fix your problem.

SaschaHenning, 2022-02-07:
Sorry, still no change.

```
2022-02-07 13:08:40 DEBUG (MainThread) [custom_components.sunspec] New SunspecApi Client
2022-02-07 13:08:40 DEBUG (SyncWorker_8) [custom_components.sunspec] Client connect to IP 192.168.0.245 port 502 slave id 1 using timeout 60
2022-02-07 13:08:40 DEBUG (SyncWorker_8) [custom_components.sunspec] Client connected, perform initial scan
2022-02-07 13:08:40 ERROR (MainThread) [custom_components.sunspec] Failed to connect to host 192.168.0.245:502 slave 1 - Repeating group count not consistent with model length for model 803,model repeating len = 6, model repeating group len = 32
```

siku2, 2023-08-24:
I recently decided to investigate this a bit, because the S10 actually exposes _more_ information in the sunspec mode than in the proprietary E3DC mode.

As it turns out, sunspec isn't really great at backwards-compatibility. The S10 uses the spec from 2016 and the models for 801, 802, and 803 have changed significantly (and in a backwards-incompatible way) since then.
If you manually change the pysunspec2's models submodule all the way back to a commit in 2016 the S10 works out of the box.
I don't think sunspec really considers this a bug or even something that should be supported. The library isn't designed to dynamically swap these model definitions, so I really don't see a way for this to be fixed.

I might create a separate integration specifically for the S10's sunspec mode in the future to handle this.

enoch85, 2026-11-13:
@SaschaHenning Just made a release which should help with this. 

Can you please try out the latest release and report back? 🙏🏼

enoch85, 2026-11-15:
> I might create a separate integration specifically for the S10's sunspec mode in the future to handle this.

And this is why I think we should support it @CJNE. It was in Sunspec, they removed it because they don't care - but we have users that expect things to work, so we should keep backwards compatibility IMHO.