CJNE/ha-sunspec #66: Wrong values with Fronius inverter

QbaF, 2021-11-19:
[As described in conersation](https://community.home-assistant.io/t/custom-component-generic-sunspec-modbus-tcp-monitoring-inverter-meter-etc/312739/41?u=qbaf).

CJNE, 2022-02-07:
Not sure if this is still relevant, feel free to re-open if so, and add some more details.

closed 2022-02-07 by CJNE

QbaF, 2022-04-06:
Oh, I missed somehow this answer, sorry. 
From my point of view still nothing changed. I do have still the wrong data in most sensors.

This is example as used on community forum mentioned in first post. That indicate stable 11'668 Watts for few months. And I do have 8.2kWp inverter.

[image]

What details I can add? I will be more than happy to assist.

QbaF, 2022-04-07:
@CJNE I am, most probably, unable to reopen the issue which was closed by repo collaborator.