CJNE/ha-sunspec #86: Serial support

CJNE, 2022-02-08:
User request for support for serial connection in addition to TCP

Add a configuration option for connection type, TCP or serial with an additional step for connection details.

jaapiyo, 2023-10-15:
Hey, I'm currently interested in this feature and I've tried to hack together some edits in a copy of your integration in order to use the modbus RTU over USB serial. However I can't seem to get it to work, lacking experience with asynchronous Python and knowledge about writing custom Home Assistant integrations.

Would it be possible for you to help me out with adding this functionality? 

I changed the TCP client for the RTU one in the following functon (api.py)
```python
def modbus_connect(self, config=None):

        # client = modbus_client.SunSpecModbusClientDeviceTCP(
        #     slave_id=use_config.slave_id,
        #     ipaddr=use_config.host,
        #     ipport=use_config.port,
        #     timeout=TIMEOUT,
        # )
        client = modbus_client.SunSpecModbusClientDeviceRTU(slave_id=use_config.slave_id, name=use_config.host)

```
Reusing the `config.host` variable for the device name. 

Unfortunately hacking this is not as simple as I thought and I keep getting the following error:

[image]

I have verified that my hardware is working and I can get Sunspec data from my SolarEdge inverter on my laptop (although using a different module called "solaredge_modbus"):

```python
import solaredge_modbus

inverter = solaredge_modbus.Inverter(device="COM6", baud=115200)

inverter.connect()

if inverter.connected():
    print(inverter.read_all())
else:
    print("not connected")

inverter.disconnect()
```

Returning the following data: 
```json
{
    "c_id": "SunS",
    "c_did": 1,
    "c_length": 65,
    "c_manufacturer": "SolarEdge",
    "c_model": "SE6K-RW0TEBEN4",
    "c_version": "0004.0014.0228",
    "c_serialnumber": "********",
    "c_deviceaddress": 1,
    "c_sunspec_did": 103,
    "c_sunspec_length": 50,
    "current": 1751,
    "l1_current": 584,
    "l2_current": 584,
    "l3_current": 582,
    "current_scale": -2,
    "l1_voltage": 4155,
    "l2_voltage": 4109,
    "l3_voltage": 4100,
    "l1n_voltage": 2367,
    "l2n_voltage": 2388,
    "l3n_voltage": 2385,
    "voltage_scale": -1,
    "power_ac": 4163,
    "power_ac_scale": 0,
    "frequency": 5001,
    "frequency_scale": -2,
    "power_apparent": 4168,
    "power_apparent_scale": 0,
    "power_reactive": -20123,
    "power_reactive_scale": -2,
    "power_factor": -9988,
    "power_factor_scale": -2,
    "energy_total": 7459710,
    "energy_total_scale": 0,
    "current_dc": 5629,
    "current_dc_scale": -3,
    "voltage_dc": 7509,
    "voltage_dc_scale": -1,
    "power_dc": 4227,
    "power_dc_scale": 0,
    "temperature": 3465,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0,
    "rrcr_state": 0,
    "active_power_limit": 100,
    "cosphi": 0,
    "commit_power_control_settings": 0,
    "restore_power_control_default_settings": 0,
    "reactive_power_config": 0,
    "reactive_power_response_time": 200,
    "advanced_power_control_enable": 1,
    "export_control_mode": 0,
    "export_control_limit_mode": 0,
    "export_control_site_limit": 0
}
```

enoch85, 2026-11-13:
@cjne I looked into what serial support would take. pysunspec2 already has a Modbus RTU client (`SunSpecModbusClientDeviceRTU`), so the work is in the integration: a connection type step in the setup flow, port, baud rate and parity fields, an RTU branch next to the TCP-only port probe and reconnect, and tests with a virtual serial port. I have no serial hardware to test on. What do you think: should we add it, or stay TCP-only and point people to a Modbus TCP gateway?

closed 2026-11-13 by enoch85