CJNE/ha-sunspec #88: Sensor is count wrong direction.

mbuhansen, 2022-02-28:
## Version of the custom_component
v0.0.20

## Configuration

```yaml
Add your logs here.

state_class: total_increasing
integration: sunspec
sunspec_key: TotWhExp
label: Total Watt-hours Exported
unit_of_measurement: Wh
device_class: energy
icon: mdi:solar-panel
friendly_name: Ac_meter Total Watt-hours Exported

## Describe the bug

A clear and concise description of what the bug is.

Hi i use this sunspec intregration, i have problem with one sensor total watt hours exported, the value in HA is count down, In my kostal smart meter the value is count up.
When i add the sensor to energy dashboard in HA it can't rister the feed in power correct because the sensor is count down the value. 
Is it possible to chenge so the value is count UP.

Add your logs here.

```

CJNE, 2022-12-12:
This sounds odd, do you think the value returned is a negative number?
What inverter model is this?

jeffry1986, 2024-10-26:
I have the same with my KOSTAL smart Meter Energy G1

which details I must share? 
exported energy will be displayed as negative value which HA interpret wrong…

enoch85, 2026-11-13:
This is #249: the Kostal Smart Energy Meter sent Total Watt-hours Exported as a signed number, which reads as a huge value that counts down. #249 reports that Kostal fixed it in the meter's April 2025 firmware. cc @cjne

enoch85, 2026-11-13:
Closing this one, if it pops up again, please file another issue. 🙏🏼

closed 2026-11-13 by enoch85