CJNE/ha-sunspec #91: Models Reconfiguration

wrongisthenewright, 2022-03-07:
## Version of the custom_component

0.0.20

## Configuration
[image]

HA 2022.3.2

Add your logs here.

No new log line is produced

## Describe the bug

After initial configuration I wanted to add meter specific values so I tried to add the 65233 model (Configuration>integrations>sunspec>configure) but the config gui gives an error of malformed input on already configured model

[image]
d

Mark481971, 2022-04-23:
Ihave the same problems in the case I want to change the "Scan Interval".

blankey85, 2022-06-11:
I also am having the same issue.  I have a secondary question also.  I have a WattNode meter that is SunSpec certified and shows I need models 1, 203, 213 for it to work.  However, in my dropdown for the read models, I can only choose models that start with a "1" (ie 1, 160, 11, 126, etc.). How do I go about adding those models?

teampyt, 2022-09-09:
Same here, I am unable to change e.g. the poling interval... Can't be much, right?

commit 5247fbb4 (2022-09-22): Fixed modles multi select options, fixes #91
closed 2022-09-22 by CJNE