CJNE/ha-sunspec #94: Solar inverter is recognized but unavailable after

hofvanta, 2022-03-08:
HA - SunSpec integration with APsystem ECU_R_PRO (TCP/IP) is not working well. When you start the integration, the ECU_R_PRO is discovere (host-ip) and also the red-modules (1, 101, 111,123). After the inverter is imported and device overview will show the (value) details. But after several seconds the device and entities becomes unavailable. 
The ECU_R_PRO is configured with multiple inverters which assigned an unique address in the modbus configuration. 
When additional inverters are added the same behavior takes places (see the values ones).
Deletion of the inverter, will block reactivation again for the same address until HA is restarted.

Please advise me how to generate the (debug) logs to investigate the problem and to help to develop the SunSpec integration for APsystems.

nielssteenvoorde, 2022-03-18:
Same dit me with a ECU-C. Evertthing Comes up and than stops working

dewi-ny-je, 2022-04-04:
Same experience with me (ECU-R-PRO).

I wonder whether a physical Modbus connection would be more stable, since cheap modbus-wifi bridges are available and work well (EW-11 from Hi-flying, I have one for my heater and it works well).

hofvanta, 2022-04-05:
I disbled/enabled each inverter with different kind of models one by one. With only one inverter and model 111, I get:
2022-04-05 19:20:01 WARNING (MainThread) [homeassistant.config_entries] Config entry '192.168.1.15:502:2' for sunspec integration not ready yet: None; Retrying in background
2022-04-05 19:20:12 WARNING (MainThread) [custom_components.sunspec] Response timeout
2022-04-05 19:20:37 WARNING (MainThread) [custom_components.sunspec] Response timeout
2022-04-05 19:20:37 ERROR (MainThread) [custom_components.sunspec] Error fetching sunspec data: 
2022-04-05 19:20:42 WARNING (MainThread) [custom_components.sunspec] Response timeout
2022-04-05 19:21:07 WARNING (MainThread) [custom_components.sunspec] Response timeout
2022-04-05 19:21:12 WARNING (MainThread) [custom_components.sunspec] Response timeout
2022-04-05 19:21:37 WARNING (MainThread) [custom_components.sunspec] Response timeout
2022-04-05 19:21:42 WARNING (MainThread) [custom_components.sunspec] Response timeout
2022-04-05 19:22:07 WARNING (MainThread) [custom_components.sunspec] Response timeout
2022-04-05 19:22:12 WARNING (MainThread) [custom_components.sunspec] Response timeout
The last messages repeated.

nielssteenvoorde, 2022-07-07:
In  ECU-C Installation/User Manual page 23:
**For three-phase Microinverter: Please refer to the document**
https://certifications.sunspec.org/PICS/Altenergy_Power_System_Inc.___three_phase_microinverter_11-12.xlsx
**For single-phase Microinverter: Please refer to the document:**
https://certifications.sunspec.org/PICS/Altenergy_Power_System_Inc.___single_phase_MicroinverterAPpcs_11-12.xlsx

There area tabs called: 1, 103, 113 and 123 with some instruction and Adress values I don't get.
Now running:
Home Assistant 2022.7.0
Supervisor 2022.07.0
Operating System 8.2
Frontend-versie: 20220706.0 - latest

It looks like something closes when adding into HA. There are some values in screen and after a couple of seconds the values in screen are all empty. Anyone any idea?

hofvanta, 2022-07-07:
Try so change the interval back to 5s. That worked for me. But it stressed the ECU after some time.

CJNE, 2022-12-12:
I suspect that inverter hardware is a bit weak and is struggling with serving the modbus requests. 
MIght be worth checking if there is a firmware update that might fix this?

closed 2026-11-12 by enoch85