CJNE/ha-sunspec #99: Data for ECU-R-PRO not matching native interface

dewi-ny-je, 2022-04-01:
The values reported
1) don't match the values from the native web interface
2) the values for both solar panels of each inverter are always the same, even if the native interface offers independent readings.

See screenshots

[image]

[image]

(the ones from 3 to 6 are not posted for conciseness)

Original interface

[image]

jeroenkl, 2022-06-01:
same issue, waiting for support

MartijnVdS, 2022-11-24:
ECU-R exposes the inverters, not the separate panels.

However, it does expose the inverters twice, once with integer data + scale factors in model 101/102/103 (single/split/multi phase), once as floats (Sunspec models 111/112/113).

I only have one inverter, so I don't know if each inverter shows up separately, or if it's all added up by the ECU-R into one big "model" object.

SenH, 2022-11-24:
You need to assign each inverter a different Modbus address on the ECU-R but there is no separate panel data indeed.
[image]

closed 2022-11-24 by dewi-ny-je