CJNE/ha-sunspec #116: Ability to change settings (write to modbus register)

okoohler, 2022-06-08:
Great app, works fine with my Fronius Gen24 with battery and smart meter. 
I'm really new to home assistant so it might just be me missing something, but I would really like to have the ability to change settings through this integration, i.e write to modbus registers in order to control charging/discharging of my batteries from/to grid. Inspiration can be found here: https://github.com/wlcrs/huawei_solar 

BR Oscar

matzkarlzzon, 2022-09-15:
Hello I stumbel into this comment and are in the same ide to controll my fronius gen24 to optimize when to load and when to charge.

I thought I know what to do by Writing the register OutWrte,InWrte perhaps StorCtl_Mod with the built in modbus function.
and the datra is change but it don´t regelat acording to what I think it should like charging the battery when out is set to -600 in to 1000 and stor to 1. inspired by https://community.home-assistant.io/t/integration-of-a-fronius-symo-gen-24-plus-inverter-via-modbus-tcp/264577/40 but fail so far ;(

Have you succede with any controll

teampyt, 2022-09-18:
Hi there, 
yes, I would like this feature as well. I could "modulate" charging for energy management. 
E.g. use some electricity for a water heater before the battery is full...

enoch85, 2026-11-14:
What do you think @CJNE?

I say adding write support is out of scope, but I might change my mind. :)

CJNE, 2026-11-14:
I can see why this would be useful, and would be a natural fit for an integration that is aimed at being a bridge between HA and SunSpec devices. I don't have the hardware to use it on but others might, i would think being able to control for example a battery would be very useful if you want HA to automate it.

enoch85, 2026-11-14:
> I can see why this would be useful, and would be a natural fit for an integration that is aimed at being a bridge between HA and SunSpec devices. I don't have the hardware to use it on but others [...]

I do lots of writing through NODE-Red already, so I don't _need_ it in Sunspec, but I agree it would be useful. So I guess I changed my mind already. 😆 

Let's aim for adding it then. I can test on my own Fronius + BYD setup.

enoch85, 2026-11-15:
This is now in main. Please feel free to test the beta and report back as new issues if you find anything strange.

You can find the beta here: https://github.com/enoch85/ha-sunspec/releases/tag/0.1.0-beta.1

closed 2026-11-15 by enoch85