CJNE/ha-sunspec #126: Configuration.yaml addition needed?

Flashlab2, 2022-07-20:
I run HA on Docker on Synology. I manually added the custom component using the instructions but SunSpec can not be found in the Integrations after restart. I see that with other manually added custom components you need to add it to the configuration.yaml file.

Does this need to be done here too? If so how?

Flashlab2, 2022-08-16:
Answer is yes. I added sunspec: in the configuration.yaml and then it works.

Rincewind76, 2022-08-24:
Could you please elaborate? I added a line 
sunspec:
in the configuration.yaml - but still nothing visible.

Flashlab2, 2022-08-26:
What I did is I copied all files in the custom components folder as is instructed in the ha-sunspec documentation. After the restart of HA though Sunspec didn't become available in the Integrations tab. For that I also needed to add the line sunspec: in the configuration.yaml file. Then it became available after restart of the HA server. Also make sure you refresh your browser.

CJNE, 2022-11-23:
Closing this as I don't think there is an issue with the integration. Installation using HACS is easier and will require no editing of yaml.

closed 2022-11-23 by CJNE