CJNE/ha-sunspec #138: Fronius temporariliy "unabailabe" data on MPPTs 2 and 3, used for the battery.

teampyt, 2022-09-18:
Hi,

I use this ha integration ver. 0.20 with my Fronius Symo Gen24 plus. 
All (but changing configuration, already an open issue) works fine.
But, due to the behaviour of the battery handling in register map160 by fronius, I get intermittent "unavailable" as state of battery "mppt module 2 Lifetime Energy" and "mppt module 3 Lifetime Energy". (MPPT 2 and 3 are being used for the battery charge and discharge) 
This seems to be due to the fact, that fronius sets these (as well as the power for the MPPTs) to 0. Fronius hotline claims that this behaviour would be correct, according sunspec definitions...
The ha-sunspec integration seems to handle that case not properly. In my opinion that needs to be changed, so that the data does not get "unavailable". Maybe the last correct value (non 0) could be given to the entity, whenever fronius sends "0" to indicate that the battery is being either "being charged" or "being discharged" or "idle".
Maybe the "input ID string" could  be used to identify that a battery is connected, to adapt the output for MPPT 2 and 3 values in case of reading 0's? 

Thank you all for your great work with this integration!!! 

teamvc

[image]

[image]

teampyt, 2022-09-25:
Update: 
I think just not sending the value 0 on MPPT2 and MPPT3 should do the trick, to not invalidate the entity by becoming "unknown", right?

teampyt, 2022-09-26:
Update: 
I think just not sending the value 0 on MPPT2 and MPPT3 should do the trick, to not invalidate the entity by becoming "unknown", right? 
This is the result of it. Data is not consistant.
[image]

teampyt, 2022-10-01:
Sorry, accidentally closed. It is not solved....

teampyt, 2022-11-15:
Update with V0.21.
If I restart the integration after the values are non zero for mppt 2 or mppt 3 it works until the value 0 is sent by the fronius inverter as lifetime energy. So restarting twice a day (after the value is above 0) circumvents the problem  (for half a day/night). Maybe that is helping to solve the issue?
teampyt

teampyt, 2022-12-12:
Thank you for the new release 0.22. 
I deleted ha-sunspec, installed 0.22 and reconfigured it. 
Then I get initially 56 entities in #160, the one that is currently is 0, is missing, which is normal.
Then the situation changes (switch from charging to discharging or vice versa) and the other MPPT lifetime energy gets 0 and the one that was 0, has now a valid value. Then the “new” non-0 mppt lifetime is NOT found nor later updated. If I restart ha-sunspec the count of entities changes to 57 and the other mppt is found.
The now 0 mppt lifetime is send out as unavailable again, so no change for me in 0.22, which was the core problem described in issue #138.
Issue #138 was closed, but it is not solved, sorry. 
So I reopen it. 
If I can assist in any way, please let me know.
teampty

teampyt, 2023-08-30:
Fronius annouced to fix it in their firmware in the future...

closed 2023-08-30 by teampyt

Cowtschi, 2024-05-16:
Hi, i have the same problem, not solved by fronius or SunSpec-integration. Do you have a solution?

boern99, 2024-12-14:
Looks like this is solved in an Firmware Update in the Middle of 2024. At least the Values are available since Month on my setup.

teampyt, 2024-12-14:
I confirm, solved by fronius.

commit a282c9bd (2022-11-25): Return last known value for energy sensors, instead of none, fixes #138