CJNE/ha-sunspec #187: Issue after Updating Homeassistant to 2023.05

christiankropf, 2023-05-05:
After Updating the entity "Solaredge inverter Watts" does not refresh properly. After reloading the integration it works for some minutes.

Tatanu, 2023-05-05:
Same issue here !

koehlerhh, 2023-05-05:
Same here. Worked properly with Kaco Blueplanet inverter until update to 2023.5. Since then data refresh from inverter obviously dysfunctional .

masieri, 2023-05-05:
Hi all, same problem for me; SMA sunnyboy 6.0 + Sunny boy storage 5.0; sunspec data stop refresh; data refresh only once after a reboot; problem started with version 2023.05. 
If you want some help please tell me for sending other informations, log and so on.

en2te, 2023-05-06:
Same here with Kaco bp 10.0 NX3 M2. No refreshing after initial update.

shymata, 2023-05-06:
Hi everybody.

I can confirm the issue :  no refresh since 2023.05 update.
I have to do it manually.

Through : SE7K-RW0TEBEN4 from SolarEdge.

If it's possible doing something in the code locally, waiting an update, that's ok.

Thanks many times for the work and the answers. ;)

shymata, 2023-05-07:
> Hi everybody.
> 
> I can confirm the issue : no refresh since 2023.05 update. I have to do it manually.
> 
> Through : SE7K-RW0TEBEN4 from SolarEdge.
> 
> If it's possible doing something in the code locally, waiting an update, that's ok.
> 
> Thanks many times for the work and the answers. ;)

Ok replying to myself.

Done an automation doing that in the meanwhile.

If you're interested search more about the service : 
`homeassistant.reload_config_entry`
Using "Config entry id".
The id of the integration is in the URL, when you click on a device of the integration Sunspec. _(after config_entry=)_

thecem, 2023-05-07:
Here the log entry, may it could help.

```
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:583 
Integration: Sensor (documentation, issues) 
First occurred: 19:26:05 (18 occurrences) 
Last logged: 19:35:58

Error adding entities for domain sensor with platform sunspec
Error while setting up sunspec platform for sensor
Error adding entities for domain sensor with platform volkswagencarnet
Error while setting up volkswagencarnet platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: ''

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 455, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 731, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 846, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
ValueError: Sensor sensor.fg2410ak3l_201_ac_meter_events has device class None, state class None unit  and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value:  (<class 'str'>)

```

Grzybu123, 2023-05-07:
Same problem here.

CJNE, 2023-05-07:
Seems like HA is more picky about data types now, the integration is now handling enum data types properly so this error should be fixed in 0.0.23.

alyf80, 2023-05-08:
Confirm fixed in 0.0.23. And now we also have proper history for enums... Yay!

christiankropf, 2023-05-08:
I can confirm it is fixed in 0.0.23. Thank you for the quick fixing!

thecem, 2023-05-08:
```
A part is working, another not:

Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:561 
Integration: Sensor (documentation, issues) 
First occurred: 12:13:40 (2 occurrences) 
Last logged: 12:13:40

Error adding entities for domain sensor with platform sunspec
Error while setting up sunspec platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 455, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 731, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 846, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 561, in state
    raise ValueError(
ValueError: Sensor sensor.pikoiq42_mppt_module_0_operating_state provides state value '0', which is not in the list of options provided

```

thecem, 2023-05-08:
[image]

thecem, 2023-05-08:
@CJNE I tried to find in which file the dct events, but not really sure:
https://github.com/sunspec/models/blob/cd7767e199ef5e7909f7d6aa17dd76b865a185b9/json/model_160.json#L160

So before it was on MPPT, I do not how the value would be mapped to the option.

thecem, 2023-05-08:
[image]

CJNE, 2023-05-08:
Oh, it looks like I missed to change the value property to use the enum values, this might have caused issues still.
Version 0.0.24 is out with a fix!
If you still have problems, please acivate debug logging and see what options it says are allowed for the sensor:

Edit configuration.yaml and add debug logging for SunSpec
logger:
  ...
  logs:
    custom_components.sunspec: debug
    ...

thecem, 2023-05-09:
```
Logger: homeassistant
Source: components/sensor/__init__.py:561
First occurred: 07:16:15 (4 occurrences)
Last logged: 07:16:51

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 204, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 349, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 146, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 406, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 561, in state
    raise ValueError(
ValueError: Sensor sensor.pikoiq42_mppt_module_0_operating_state provides state value '0', which is not in the list of options provided
```
```
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:561
Integration: Sensor (documentation, issues)
First occurred: 07:16:03 (2 occurrences)
Last logged: 07:16:03

Error adding entities for domain sensor with platform sunspec
Error while setting up sunspec platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 455, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 731, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 846, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 561, in state
    raise ValueError(
ValueError: Sensor sensor.pikoiq42_mppt_module_0_operating_state provides state value '0', which is not in the list of options provided

```

```
2023-05-09 07:16:03.884 DEBUG (MainThread) [custom_components.sunspec] Created sensor for Evt in model 160 using prefix PikoIQ42: Pikoiq42 mppt Global Events uid 9da32dbf29e3d958fe5a714c4adbb7a5_Evt-160-0, device class enum unit None
2023-05-09 07:16:03.884 DEBUG (MainThread) [custom_components.sunspec] Valid options for ENUM: ['GROUND_FAULT', 'INPUT_OVER_VOLTAGE', 'RESERVED_2', 'DC_DISCONNECT', 'RESERVED_4', 'CABINET_OPEN', 'MANUAL_SHUTDOWN', 'OVER_TEMP', 'RESERVED_8', 'RESERVED_9', 'RESERVED_10', 'RESERVED_11', 'BLOWN_FUSE', 'UNDER_TEMP', 'MEMORY_LOSS', 'ARC_DETECTION', 'RESERVED_16', 'RESERVED_17', 'RESERVED_18', 'RESERVED_19', 'TEST_FAILED', 'INPUT_UNDER_VOLTAGE', 'INPUT_OVER_CURRENT', '']
2023-05-09 07:16:03.885 DEBUG (MainThread) [custom_components.sunspec] Created sensor for module:0:DCA in model 160 using prefix PikoIQ42: Pikoiq42 mppt module 0 DC Current uid 9da32dbf29e3d958fe5a714c4adbb7a5_module:0:DCA-160-0, device class current unit A
2023-05-09 07:16:03.885 DEBUG (MainThread) [custom_components.sunspec] Created sensor for module:0:DCV in model 160 using prefix PikoIQ42: Pikoiq42 mppt module 0 DC Voltage uid 9da32dbf29e3d958fe5a714c4adbb7a5_module:0:DCV-160-0, device class voltage unit V
2023-05-09 07:16:03.885 DEBUG (MainThread) [custom_components.sunspec] Created sensor for module:0:DCW in model 160 using prefix PikoIQ42: Pikoiq42 mppt module 0 DC Power uid 9da32dbf29e3d958fe5a714c4adbb7a5_module:0:DCW-160-0, device class power unit W
2023-05-09 07:16:03.885 DEBUG (MainThread) [custom_components.sunspec] Created sensor for module:0:Tms in model 160 using prefix PikoIQ42: Pikoiq42 mppt module 0 Timestamp uid 9da32dbf29e3d958fe5a714c4adbb7a5_module:0:Tms-160-0, device class None unit s
2023-05-09 07:16:03.885 DEBUG (MainThread) [custom_components.sunspec] Created sensor for module:0:Tmp in model 160 using prefix PikoIQ42: Pikoiq42 mppt module 0 Temperature uid 9da32dbf29e3d958fe5a714c4adbb7a5_module:0:Tmp-160-0, device class temperature unit °C
2023-05-09 07:16:03.886 DEBUG (MainThread) [custom_components.sunspec] Created sensor for module:0:DCSt in model 160 using prefix PikoIQ42: Pikoiq42 mppt module 0 Operating State uid 9da32dbf29e3d958fe5a714c4adbb7a5_module:0:DCSt-160-0, device class enum unit None
2023-05-09 07:16:03.886 DEBUG (MainThread) [custom_components.sunspec] Valid options for ENUM: ['OFF', 'SLEEPING', 'STARTING', 'MPPT', 'THROTTLED', 'SHUTTING_DOWN', 'FAULT', 'STANDBY', 'TEST', 'RESERVED_10', '']
[... cut ...]
2023-05-09 07:20:00.224 DEBUG (MainThread) [custom_components.sunspec] SunSpec Update data got models {203}
2023-05-09 07:20:00.224 DEBUG (MainThread) [custom_components.sunspec] Get data for model 203
2023-05-09 07:20:00.258 DEBUG (MainThread) [custom_components.sunspec] SunSpec Update data coordinator update
2023-05-09 07:20:00.258 DEBUG (MainThread) [custom_components.sunspec] Fetching models
2023-05-09 07:20:00.260 DEBUG (MainThread) [custom_components.sunspec] SunSpec Update data got models {203}
2023-05-09 07:20:00.260 DEBUG (MainThread) [custom_compo

thecem, 2023-05-09:
Let me know if you need more, the Fronius seems to be with no errors but the kostal piko.

thecem, 2023-05-14:
Do you need more details? 
The values are not updating. Only once you reload the integration or boot HA.

shingsam, 2023-05-18:
> Oh, it looks like I missed to change the value property to use the enum values, this might have caused issues still. Version 0.0.24 is out with a fix! If you still have problems, please acivate [...]

Hi, why i can't get result when last value Unavailable. Soo i need to restart HA and i can get result again.
SunSpec version 0.0.24

Thanks,

Grzybu123, 2023-07-20:
Is there a fix for the plugin? In some other plugins, the problem was in the phyton version.

enoch85, 2026-11-12:
Fixed: since 0.0.25 an enum value that is not one of the model's symbols, such as the `0` from the Kostal MPPT module, gives the state unknown instead of this error (8126364), and #384, merged after 0.0.29, reports a bitfield with several flags set instead of having it refused. cc @cjne

closed 2026-11-12 by enoch85