CJNE/ha-sunspec #201: Request different times for polling each sunspec model.

teampyt, 2023-08-30:
Request different times for polling each sunspec model.
I think reading static date, like sunspec model 120/121 should be polled at a much lower rate as model 160, which holds dynamic data.
Please generate intervals per model.
eg.
Model 120   = 60 seconds
Model 1       = 10 seconds
Model 160   = 5 seconds
...

cyberthom42, 2025-01-03:
I switched from iobroker to HA. I was used to configure the polling interval as proposed here. Are there any news about this request?

enoch85, 2026-11-13:
@cjne What would you like here? Polling each model on its own interval means a due time per model in the coordinator (or one coordinator per interval) and an interval field per enabled model in the options. It is doable, but it adds options to maintain. Should we add it, or keep one interval per device?

closed 2026-11-13 by enoch85