CJNE/ha-sunspec #202: keep entities when they vanish from SMA TriPower X12 inverters

SteffenViperKoch, 2023-09-29:
**Is your feature request related to a problem? Please describe.**
SMA Sunny TriPower X12 inverters power down their internal CPU after solar production stops. So the integrations loses their entities and HomeAssistant shows them missing instead of just no value

**Describe the solution you'd like**
I'd like to see the entities stay "alive" just without a value (if that's possible) or just with 0 value

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

reeeeevls, 2023-11-25:
I have the same problem. Does your entities show up the next time the sun is shining? Mine are lost since I "reconfigure" SunSpec.

pandersin, 2024-02-08:
Same problem with SMA TriPower STP50-41. Entities do show up in the morning if the HA has not restarted during the night.

matze338, 2024-02-09:
In the night, the entities goes unavailable.
I also got some warnings and errors in the log:
```
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.sunspec
Source: custom_components/sunspec/__init__.py:154
Integration: SunSpec (documentation, issues)
First occurred: 18:08:08 (1365 occurrences)
Last logged: 20:03:03

Modbus exception 4: addr: 40096 count: 123
```

```
Logger: homeassistant.helpers.event
Source: helpers/event.py:296
First occurred: 18:14:33 (1 occurrences)
Last logged: 18:14:33

Error while dispatching event for sensor.sunspec_dermeasureac_total_energy_injected to <Job track state_changed event {'sensor.sunspec_dermeasureac_total_energy_injected'} HassJobType.Callback <bound method TrackTemplateResultInfo._refresh of <TrackTemplateResultInfo {Template<template=({{ states('sensor.sunspec_dermeasureac_total_energy_injected') }}) renders=366>: <RenderInfo Template<template=({{ states('sensor.sunspec_dermeasureac_total_energy_injected') }}) renders=366> all_states=False all_states_lifecycle=False domains=frozenset() domains_lifecycle=frozenset() entities=frozenset({'sensor.sunspec_dermeasureac_total_energy_injected'}) rate_limit=None has_time=False exception=None is_static=False>}>>>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 669, in state
    numerical_value = float(value)  # type:ignore[arg-type]
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 296, in _async_dispatch_entity_id_event
    hass.async_run_hass_job(job, event)
  File "/usr/src/homeassistant/homeassistant/core.py", line 700, in async_run_hass_job
    hassjob.target(*args)
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1215, in _refresh
    self.hass.async_run_hass_job(self._job, event, updates)
  File "/usr/src/homeassistant/homeassistant/core.py", line 700, in async_run_hass_job
    hassjob.target(*args)
  File "/usr/src/homeassistant/homeassistant/components/template/template_entity.py", line 434, in _handle_results
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 989, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1110, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1047, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 995, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 671, in state
    raise ValueError(
ValueError: Sensor sensor.strom_produktion_energie has device class 'energy', state class 'total' unit 'Wh' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'unavailable' (<class 'str'>)

```

And
```
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.sunspec
Source: helpers/update_coordinator.py:345
Integration: SunSpec (documentation, issues)
First occurred: 18:14:33 (1 occurrences)
Last logged: 18:14:33

Error fetching sunspec data:
```
[image]

In the morning the integrations works again

Syonis, 2024-03-18:
Same here with my Symo 7.0-3-M while the Gen24 is still available. 
But the reason here is, that the Symo is going down in the night and the Gen24 not. 
"I'd like to see the entities stay "alive" just without a value (if that's possible) or just with 0 value"
That would be a nice one.

michaelpiron, 2024-05-04:
Same issue here. Entities become unavailable during the night, and don't become available again the next day.
Strangely enough, the total Wh counter and inverter state become available again, all the other entities remain unavailable.
Hardware: SMA sunny boy SB 3000TL-21

[image]

hilman2, 2026-08-19:
For anyone still following this thread: the behaviour you are asking for is implemented in [ha-sunspec2](https://github.com/hilman2/ha-sunspec2), a fork of this integration, and it takes two separate mechanisms rather than one.

**Keeping the entities alive across the night.** The entity's `available` property does not follow `last_update_success`. It stays `True` while the coordinator's consecutive-failure count is below a tolerance (5 cycles at a 30 s interval, so about two and a half minutes), so the sensors keep reporting their last good value through a short outage instead of going unavailable immediately.

There is a subtlety worth knowing if anyone ports this: HA's `DataUpdateCoordinator._async_refresh` early-returns once failures repeat, so the listeners never fire again and the entity would stay frozen on the stale value *forever* once the threshold is crossed. The tolerance only works if you call `async_update_listeners()` yourself at the moment the counter crosses it.

**The model disappearing entirely**, which is the TriPower X12 behaviour @matze338 and @pandersin describe: each model that was seen once and then stops appearing in the scan is tracked with the time it went missing, and after ten minutes of continuous absence a Repairs issue suggests removing that device. It does not delete anything by itself, because a model that vanishes at dusk and returns at dawn is not a problem to be fixed.

Both have been in place for a while and this issue is what they were written against.

enoch85, 2026-11-13:
@cjne Since 0.0.30 (#391) the sensors come back by themselves in the morning, also after a restart at night. What is left here is keeping them available while the inverter sleeps and the polls fail. Home Assistant's guidance is to mark entities unavailable when the device cannot be reached, so I would leave that as it is. Do you agree, or would you rather keep the last values through the night?

closed 2026-11-13 by enoch85