CJNE/ha-sunspec #228: no connection to TRIPOWER X 25

Bingo2023, 2023-10-23:
## Version of the custom_component
Version 0.0.25
-->
[image]

## Describe the bug
SMA TRIPOWER X 25 can not be added. Modbus is enabled and Modbus-data can be seen in Nodered.
Firmware-Version: 03.02.08.R

Add your logs here.
[image]

```

Bingo2023, 2023-10-24:
after disconnecting the Ethernet cable for some other installation work, HA found the SMA device.
Reset etc was not needed, just the ethernet cable....

enoch85, 2026-11-13:
The `'H' format` error comes from the scan: the inverter's SunSpec map ran past the last Modbus address, 65535, so the scan could not finish and nothing past that point can be read. In #244, several Sunny Tripower X owners found that restarting the inverter fixed it. #398 closes the connection that each failed attempt left open. cc @cjne

closed 2026-11-13 by enoch85