CJNE/ha-sunspec #244: Failed to connect to SMA SUNNY TRIPOWER X 15

mxtrck, 2024-01-07:
## Version of the custom_component
Version 0.0.25

## Configuration

[image]

## Describe the bug

Failed to connect to SMA SUNNY TRIPOWER X 15 
Can't add the Inverter.  
Logs says 'H' format requires 0 <= number <= 65535

## Debug log

```text

2024-01-07 01:31:13.627 ERROR (MainThread) [custom_components.sunspec] Failed to connect to host 10.7.20.105:502 slave 126 - 'H' format requires 0 <= number <= 65535

```

littleyoda, 2024-01-21:
Just to check:
Have you activated modbus in the Tripower X?
Can you access the inverter via https://10.7.20.105

290989, 2024-01-22:
I can't connect to the same device after update of home assistant core.
Any Ideas? Have the same message

Makrokaba, 2024-01-23:
Same problem with our SMA Tripower X 12. It used to work fine before. 
Modbus server is switched on in the Tripower X and its webinterface can be accessed.
I didn't notice at first because our roof was covered with snow for a while...
Apart frome the new HA version there has been a firmware update from SMA to version 03.02.11.R in the time between.

Ropdepil, 2024-01-24:
I have the same problem. I have a Tripower X15. Before that everything was going wonderfully. No connection since today. Homeassistant update was not carried out. I'm currently running 2024.1.3

290989, 2024-01-24:
> I can't connect to the same device after update of home assistant core. Any Ideas? Have the same message

I have restarted my inverter. After that, everything worked again (since 2 days)

Ropdepil, 2024-01-24:
> > I can't connect to the same device after update of home assistant core. Any Ideas? Have the same message
> 
> I have restarted my inverter. After that, everything worked again (since 2 days) I have a Tripower X15

Did you take off AC and DC?

Ropdepil, 2024-01-24:
> > > > I can't connect to the same device after update of home assistant core. Any Ideas? Have the same message
> > > 
> > > 
> > > I have restarted my inverter. After that, everything worked again (since 2 days) I have a Tripower X15
> > 
> > 
> > Did you take off AC and DC?
> 
> I have shut off the fuse and wait 5 minutes. Than I turned on again and it started to working again
Unfortunately it didn't help in my case

mikezieg, 2024-01-25:
Same here with Tripower X25:  from one moment to the next the tcp-connection stopped working. Connections with sunny-portal okay, sunny explorer as well. But i am not able to establish a connection from HA via SunSpec Modbus any more

mikezieg, 2024-01-25:
Second time shutting down the tripower x,  this time some minutes more - it worked!

290989, 2024-01-25:
> Second time shutting down the tripower x, this time some minutes more - it worked!

Very nice 👍🏻

BBrakhahn, 2024-02-04:
I still receive "DER AC Measurement" via SunSpec 701 but no more information via SunSpec 714 (DER DC Measurement). Modbus TCP server is active in SMA inverter obviously but i can not reach it with HASS SunSpec when i try to reconfigure. Even in a new HASS instance i can not reach it via SunSpec. But it must be reachable some how since i still receive some information from the old config.

 Sunny Tripower X 20 with firmware version 03.02.11.R.
Home Assistant: 
- Core: 2024.1.6
- Supervisor: 2023.12.1
- Operating System: 11.4
- Frontend: 20240104.0
 
I would relate the problem to the newest firmware by SMA but there were also some HASS updates since i realized missing data in SunSpec. 

Reboots in any way wont change the situation for me. 
- Reboot initiated by frontend
- Reboot initiated by fuse
- Shutdown by fuse and wait few hours before starting up again

Somewhere i have read about reconnecting LAN may help. Not for me.

Just showing my case. Maybe some day there will a solution. I am also in contact with SMA and if i get hint what to do i will report.

redfusion83, 2024-03-07:
Today Sunspec lost connection to my X 15 for unknown reasons. I didn't change anything in the configuration, and the X 15 also has the same IP as before, but Sunspec is not able to connect.

littleyoda, 2024-03-07:
> Today Sunspec lost connection to my X 15 for unknown reasons. I didn't change anything in the configuration, and the X 15 also has the same IP as before, but Sunspec is not able to connect.

There are reports in various forums that Sunsepc only worked after the last firmware update once the inverter had been restarted.

Ben-x86, 2024-03-07:
Restarting resolved it for me but ... I cant tell which one of the thousends xD

As I remeber correct I updated my whole HA instance including add-ons and integrations. When the inverter produced power I forced a restart from the local sma dashboard. After that the missing information came back with HA sunspec.

Auto update is now deactivated in SMA devices. I'm not happy with that but I want my data.

redfusion83, 2024-03-08:
> There are reports in various forums that Sunsepc only worked after the last firmware update once the inverter had been restarted.
Thanks for that, very helpful! Rebooting the X15 from the local interface restored Sunspec's connection, but it did not report all the data. I also had to reload the integration to fix this. Now everything works again.

However, there was no firmware update before the failure. The last update to 03.06.15.R was a week and a half ago. The only update that could have affected the integration was the latest HA core update to 2024.3.0, which I did the same day, but I can't say if that caused the disconnection. I'm now going to monitor certain entities so that I get a notification if this happens again.

dana-se, 2025-05-09:
Can someone explain to a beginner how to restart X15 from the local interface

Thanks

enoch85, 2026-11-13:
This is #228, the same `'H' format` error: the inverter's SunSpec map ran past the last Modbus address, 65535, which the scan cannot get past. #398 closes the connection that each failed attempt left open. cc @cjne

closed 2026-11-13 by enoch85