CJNE/ha-sunspec #245: Many enteties are no longer supported adter upgrade to HAOS 2024.1.2

enoch85, 2024-01-11:
## Version of the custom_component

Sunspec: [V0.0.25](https://github.com/CJNE/ha-sunspec/releases/tag/0.0.25)

HAOS:
Core: 2024.1.2
Supervisor: 2023.12.0
Operating System: 11.4
Frontend: 20240104.0

## Configuration

```yaml
configurration.yaml:
# Modbus
modbus: !include modbus.yaml

modbus.yaml:
# Fronius Gen24
# modbus:
  - type: tcp
    host: 10.10.10.4
    port: 502
    name: gen24

    delay: 1
    message_wait_milliseconds: 30
    timeout: 1
```

## Describe the bug

After upgrading to HAOS 2024.1.2 I more than half of my entities are gone. Was working perfectly fine before.

## Debug log

```text

Logger: homeassistant.components.sensor
Source: helpers/deprecation.py:205
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 19:56:24 (12 occurrences)
Last logged: 19:56:25

DEVICE_CLASS_ENERGY was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.ENERGY instead, please create a bug report at https://github.com/cjne/ha-sunspec/issues
DEVICE_CLASS_TEMPERATURE was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.TEMPERATURE instead, please create a bug report at https://github.com/cjne/ha-sunspec/issues
DEVICE_CLASS_VOLTAGE was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.VOLTAGE instead, please create a bug report at https://github.com/cjne/ha-sunspec/issues
STATE_CLASS_MEASUREMENT was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.MEASUREMENT instead, please create a bug report at https://github.com/cjne/ha-sunspec/issues
STATE_CLASS_TOTAL_INCREASING was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.TOTAL_INCREASING instead, please create a bug report at https://github.com/cjne/ha-sunspec/issues

```

[image]
[image]

[image]
[image]
[image]
[image]

enoch85, 2024-01-11:
Also pinging @Nesurion on this.

enoch85, 2024-01-11:
A bit embarrassing, but I just noticed that all the models weren't ticked, now it's all back except 3 entities. I can breath agai, peew.

Anyway, keeping open for the other issues.

CJNE, 2024-08-18:
I'm closing this, create a new issue if there are other problems :)

closed 2024-08-18 by CJNE