CJNE/ha-sunspec #249: `ac_meter Total Watt-hours Exported` sign issue

DooMMasteR, 2024-02-04:
## Version of the custom_component
0.025
## Configuration
[image]

## Describe the bug
`ac_meter Total Watt-hours Exported` is a huge value, slowly lowering whenever electricity is exported
[image]

dp2008, 2024-03-19:
Have the same problem with Kostal Smart Energy Meter
there seems to be a fix for openHAB but nothing for HA:
https://community.openhab.org/t/solved-wrong-values-with-modbus-and-kostal-smart-energy-meter/99785

see last comment form Oct 2022:
the Kostal SEM reports Exported Real Energy as signed int32 instead of unsigned int32, which makes it report bad values.
Same happens in Home Assistant and with the official Sunspec verification tools.

DooMMasteR, 2024-03-19:
Yeah I already tried to get KOSTAL to fix the issue in firmware, but they are super shitty at supporting their products.

guidoffm, 2024-07-21:
I have the same problem with the Kostal meter

Canonip, 2025-03-25:
My workaround of Kostals shoddy programming is the following:
`total_increasing` is important, otherwise it won't show up in the energy dashboard

    template:
      - sensor:
          - name: "Ksem ac_meter Total Watt-hours Exported Corrected"
            unique_id: "ksem_ac_meter_total_watt_hours_exported_correct"
            device_class: energy
            state_class: total_increasing
            unit_of_measurement: Wh
            state: >
              {% set value = states('sensor.ksem_ac_meter_total_watt_hours_exported') | int %}
              {{ 4294967296 - value }}

DooMMasteR, 2025-03-25:
Yeah, I am really disappointed that they still have not fixed this fundamental issue.

I am now using the readout of a Deye inverter that we also added to our installation, that has a separate rs485 meter.

closed 2025-06-02 by DooMMasteR