CJNE/ha-sunspec #257: Question: is it possible to read modbus data over wifi?

UrZdcw9, 2024-02-18:
hi there @CJNE , 

thanks for this integration :).  i have been using it very successfully for nearly two years now.

my solar edge inverter is currently using a wired connection to my router.  however, i had to make a change to this and connect the solar edge inverter  to my wifi instead - eveything was configured to remain the same (IP address, modbus ports etc).

in doing so, i was not able to read the modbus data in home assistant.  so i reverted back to the wired connection.

is this expected behaviour for all inverters or perhaps simply a limitation of my own?

many thanks.

CJNE, 2024-02-18:
Hi there!
This is a bit hard to tell, from the home assistant side of things it shouldn't make a difference wether the connection is wired or WiFi.
Perhaps your home assistant box don't have access to the WiFi?
Or maybe the inverter doesn't support modbus over the WiFI connection, that's also a possibility.

UrZdcw9, 2024-02-19:
Thanks for the response. I think it's a limitation of the inverter. Will close case.

closed 2024-02-19 by UrZdcw9