CJNE/ha-sunspec #264: Deprecation warning since 2024.5.0b0

gieljnssns, 2024-04-25:
## Version of the custom_component

v0.0.25
## Configuration

```yaml
2024-04-25 08:20:40.735 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'sunspec' calls `async_add_job`, which is deprecated and will be removed in Home Assistant 2025.4; Please review https://developers.home-assistant.io/blog/2024/03/13/deprecate_add_run_job for replacement options at custom_components/sunspec/__init__.py, line 146: self.hass.async_add_job(, please create a bug report at https://github.com/cjne/ha-sunspec/issues
```

## Describe the bug

It reports a deprecation.

## Debug log

```text

Add your logs here.

```

01815, 2024-05-05:
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.sunspec
Quelle: custom_components/sunspec/config_flow.py:127
Integration: SunSpec (Dokumentation, Probleme)
Erstmals aufgetreten: 11:40:17 (3 Vorkommnisse)
Zuletzt protokolliert: 16:28:27

Failed to connect to host 10.100.1.254:502 slave 126 - Failed to connect to 10.100.1.254:502 slave id 126
Failed to connect to host 10.100.1.254:502 slave 3 - Failed to connect to 10.100.1.254:502 slave id 3

CJNE, 2024-08-18:
Fixed in next release, thanks!

closed 2024-08-18 by CJNE