CJNE/ha-sunspec #265: Another deprecation Warning

Styx85, 2024-04-27:
There are other deprecation warnings which will be removed from HA 2025.1 onwards

```
Logger: homeassistant.components.sensor
Quelle: helpers/deprecation.py:206
Integration: Sensor (Dokumentation, Probleme)
Erstmals aufgetreten: 14:15:23 (12 Vorkommnisse)
Zuletzt protokolliert: 14:15:23

DEVICE_CLASS_ENERGY was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.ENERGY instead, please report it to the author of the 'sunspec' custom integration
DEVICE_CLASS_TEMPERATURE was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.TEMPERATURE instead, please report it to the author of the 'sunspec' custom integration
DEVICE_CLASS_VOLTAGE was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.VOLTAGE instead, please report it to the author of the 'sunspec' custom integration
STATE_CLASS_MEASUREMENT was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.MEASUREMENT instead, please report it to the author of the 'sunspec' custom integration
STATE_CLASS_TOTAL_INCREASING was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.TOTAL_INCREASING instead, please report it to the author of the 'sunspec' custom integration

```

usernr1, 2024-05-16:
Same here
```
Logger: homeassistant.components.sensor
Quelle: helpers/deprecation.py:206
Integration: Sensor ([Dokumentation](https://www.home-assistant.io/integrations/sensor), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
Erstmals aufgetreten: 8. Mai 2024 um 19:01:50 (12 Vorkommnisse)
Zuletzt protokolliert: 8. Mai 2024 um 19:01:50

DEVICE_CLASS_ENERGY was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.ENERGY instead, please report it to the author of the 'sunspec' custom integration
DEVICE_CLASS_TEMPERATURE was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.TEMPERATURE instead, please report it to the author of the 'sunspec' custom integration
DEVICE_CLASS_VOLTAGE was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.VOLTAGE instead, please report it to the author of the 'sunspec' custom integration
STATE_CLASS_MEASUREMENT was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.MEASUREMENT instead, please report it to the author of the 'sunspec' custom integration
STATE_CLASS_TOTAL_INCREASING was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.TOTAL_INCREASING instead, please report it to the author of the 'sunspec' custom integration
```

CJNE, 2024-08-18:
Fixed in next release, thanks!

closed 2024-08-18 by CJNE