CJNE/ha-sunspec #269: Cannot connect to SUNNY TRIPOWER 10.0 SE

danielbierstedt, 2024-05-22:
Hello,

I cannot connect to an SMA Tripower 10 and cannot figure out why. First thing maybe is that I don't understand the "slave id". I only have the "Unit ID" that is given in the webinterface.
I configure the inverter over webinterface on IP .43, thats where SunSpec received an answer. I suspect the IP and Port to be correct, but the "Slave ID" seems to be wrong.

Any help highly appreciated!

## Debug log

```text

Logger: custom_components.sunspec
Source: custom_components/sunspec/config_flow.py:127
integration: SunSpec ([documentation](https://github.com/cjne/ha-sunspec), [issues](https://github.com/cjne/ha-sunspec/issues))
First occurred: 17:03:15 (12 occurrences)
Last logged: 17:32:03

Failed to connect to host 192.168.0.170:502 slave 3 - Failed to connect to 192.168.0.170:502 slave id 3
Failed to connect to host 192.168.0.20:502 slave 3 - Failed to connect to 192.168.0.20:502 slave id 3
Failed to connect to host 192.168.0.43:502 slave 1 - Device responded - not SunSpec register map
Failed to connect to host 192.168.0.43:502 slave 126 - Failed to connect to 192.168.0.43:502 slave id 126
Failed to connect to host 192.168.0.43:502 slave 2 - Device responded - not SunSpec register map

```

Sorentino1964, 2024-05-30:
I have the same problem!

Dr-know, 2024-08-20:
i have the same problem !
is this integration using TCP or UDP ??

Modbus
You can enable the Modbus TCP server and set the TCP port and the Unit ID.
If you would like to use Modbus via UDP, you must turn on the UDP server via Device parameters and set the UDP port.

SmileyMan, 2024-08-23:
I reverted to 0.0.25 and it started working ok again.

enoch85, 2026-11-13:
This integration reads SMA's SunSpec Modbus profile, which has its own unit ID: SMA's Modbus documentation derives it from the SMA profile's unit ID plus 123, so 126 by default. Unit ID 3 is SMA's own Modbus profile, which this integration cannot read. cc @cjne

enoch85, 2026-11-14:
This log is from 0.0.24 or 0.0.25. There, `Failed to connect ... slave id 126` most likely meant the device answered at 126 but the scan stopped partway through its models: SMA inverters answer the register after their last model with an exception instead of an end marker, as #346 shows on another Sunny Tripower. Since 0.0.30 setup keeps the models found (#387), and since 0.0.31 a failed attempt closes its connection (#398), which matters because SMA inverters accept four. Unit ID 126 is the one to use. cc @cjne

closed 2026-11-14 by enoch85