CJNE/ha-sunspec #270: Sungrow SH10RT few sensors

kalaws, 2024-05-24:
Hi. Thanks for the excellent work!

I've got a Sungrow SH10RT-112. For some reason I only get two models (is that the correct term)? 1, and 101. 1 gives nothing, 101 gives 19 sensors. I've also got a smart meter and a battery, but nothing of those is reported. Is there any way to get more data through this integration that I've missed?

kalaws, 2024-05-24:
Also, for some reason the inverter is detected by the integration as "Inverter (Single Phase)", while it's actually three phase.

enoch85, 2026-11-13:
The integration offers the models in the inverter's SunSpec map, and this SH10RT serves only 1 and 101, the single-phase inverter model; there are no meter or battery models to read. Sungrow's SH-RT manual and Modbus protocol do not mention SunSpec and give meter and battery values only in Sungrow's own registers. cc @cjne

closed 2026-11-13 by enoch85