CJNE/ha-sunspec #273: Sun Spec no longer working after HA update to 2024.7.2

mash727, 2024-07-17:
## Version of the custom_component

## Configuration
Raspberry PI 4B as well es Intel NUC.

Add your logs here.
```

## Describe the bug
See HA protokoll

## Debug log

Logger: homeassistant.util.loop
Quelle: util/loop.py:84
Erstmals aufgetreten: 16. Juli 2024 um 14:36:46 (172 Vorkommnisse)
Zuletzt protokolliert: 02:10:12

Detected blocking call to open inside the event loop by custom integration 'sunspec' at custom_components/sunspec/api.py, line 159: client.scan( (offender: /usr/local/lib/python3.12/site-packages/sunspec2/mdef.py, line 375: f = open(filename)), please create a bug report at https://github.com/cjne/ha-sunspec/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/sunspec/__init__.py", line 54, in async_setup_entry await coordinator.async_config_entry_first_refresh() File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 277, in async_config_entry_first_refresh await self._async_refresh( File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh self.data = await self._async_update_data() File "/config/custom_components/sunspec/__init__.py", line 155, in _async_update_data self.api.reconnect() File "/config/custom_components/sunspec/api.py", line 133, in reconnect client = self.get_client() File "/config/custom_components/sunspec/api.py", line 97, in get_client cached = self.modbus_connect(config) File "/config/custom_components/sunspec/api.py", line 159, in modbus_connect client.scan(
Detected blocking call to open inside the event loop by custom integration 'sunspec' at custom_components/sunspec/api.py, line 159: client.scan( (offender: /usr/local/lib/python3.12/xml/etree/ElementTree.py, line 558: source = open(source, "rb")), please create a bug report at https://github.com/cjne/ha-sunspec/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/sunspec/__init__.py", line 54, in async_setup_entry await coordinator.async_config_entry_first_refresh() File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 277, in async_config_entry_first_refresh await self._async_refresh( File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh self.data = await self._async_update_data() File "/config/custom_components/sunspec/__init__.py", line 155, in _async_update_data self.api.reconnect() File "/config/custom_components/sunspec/api.py", line 133, in reconnect client = self.get_client() File "/config/custom_components/sunspec/api.py", line 97, in get_client cached = self.modbus_connect(config) File "/config/custom_components/sunspec/api.py", line 159, in modbus_connect client.scan(
Detected blocking call to sleep inside the event loop by custom integration 'sunspec' at custom_components/sunspec/api.py, line 159: client.scan( (offender: /usr/local/lib/python3.12/site-packages/sunspec2/modbus/client.py, line 298: time.sleep(delay)), please create a bug report at https://github.com/cjne/ha-sunspec/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/sunspec/__init__.py", line 54, in async_setup_entry await coordinator.async_config_entry_first_refresh() File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 277, in async_config_entry_first_refresh await self._async_refresh( File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh self.data = await self._async_update_data() File "/config/custom_components/sunspec/__init__.py", line 155, in _async_update_data self.api.reconnect() File "/config/custom_components/sunspec/api.py", line 133, in reconnect client = self.get_client() File "/config/custom_components/sunspec/api.py", line 97, in get_client cached = self.modbus_connect(config) File "/config/custom_components/sunspec/api.py", line 159, in modbus_connect client.scan(
Detected blocking call to open inside the event loop by custom integration 'solcast_solar' at custom_components/solcast_solar/solcastapi.py, line 200: with open(self._filename, 'w') as f: (offender: /config/custom_components/solcast_solar/solcastapi.py, line 200: with open(self._filename, 'w') as f:), please create a bug report at https://github.com/oziee/ha-solcast-solar/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_event [... cut ...] _run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/config/custom_components/solcast_solar/coordinator.py", line 138, in update_forecast await self.solcast.force_api_poll(True) File "/config/custom_components/solcast_solar/solcastapi.py", line 148, in force_api_poll await self.http_data(args[0]) File "/config/custom_components/solcast_solar/solcastapi.py", line 575, in http_data with open(self._filename, 'w') as f:
```

saschaludwig, 2024-07-31:
Looks like I have the same problem here with two Kostal inverters.

eckerse, 2024-07-31:
seems, there are some blocking calls used from upstream project sunspec2.
There has been a warning about this before, but since 2024.07 this will raise an error instead, blocking use.

I'm not familar enough with python, but HA developers offered some hints for fixing: https://developers.home-assistant.io/docs/asyncio_blocking_operations/
Hope this can be fixed.

saschaludwig, 2024-08-29:
The Sunspec Integration isn't working since over one month now for my two Kostal inverters.
The entities are available for one value, then after 60 seconds the entities are unavailable and it all starts over.
One value, unavailable, one value, unavailable.

Is there any way I can help with more logs?

closed 2026-11-12 by enoch85