CJNE/ha-sunspec #274: SunSpec after Restart HA not available

290989, 2024-08-09:
Hi, I have this error after restart Home Assistant:

Einrichtung fehlgeschlagen, wird erneut versucht: Caught blocking call to sleep with args (0.5,) inside the event loop by custom integration 'sunspec' at custom_components/sunspec/api.py, line 159: client.scan(. (offender: /usr/local/lib/python3.12/site-packages/sunspec2/modbus/client.py, line 258: time.sleep(delay)), please create a bug report at https://github.com/cjne/ha-sunspec/issues For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#sleep

Does anyone have an idea to fix it?

Thank you.

290989, 2024-08-12:
After a restart from the inverter, sunspec worked again for me

alphabravocharly72, 2024-08-14:
Hello, I have the same issue. After a restart from HA (eg update or other reasons) the integration shows no values. After a manually restart (sun spec) it works. It looks like during HA start the integration doesn't start as it should.

CJNE, 2024-08-18:
I think this should work better with the latest version, 0.0.26

timoverwoest, 2024-08-30:
Hi CJNE, I still seems to have the issue.

alphabravocharly72, 2024-08-30:
Same. Not better with the new version.

hilman2, 2026-08-19:
The `Caught blocking call to sleep` warning has a specific cause worth naming, because it is not really about the sleep.

`client.scan(delay=0.5)` makes pysunspec2 sleep half a second after every model it walks. That is fine in a worker thread and fatal in the event loop, and the scan runs wherever `get_client()` is called from. The fix is not to remove the delay but to make sure the whole connect-and-scan runs through `async_add_executor_job`, the same way `read_model` already does.

In [ha-sunspec2](https://github.com/hilman2/ha-sunspec2), a fork of this integration, `async_get_client()` wraps `get_client` in an executor job for exactly this reason, so the sleep blocks a worker thread and HA never sees it.

Two related things from working on the same code, in case they help anyone here:

- The scan is not needed on every poll. If the client is closed at the end of each cycle (which single-slot inverters need), the next cycle rebuilds it and rescans the whole model tree, which on a 20-model inverter is 41 Modbus round trips plus 20 pacing sleeps, twice a minute. Caching the model layout and validating it with one read cut a user's real update period from 60 s to 34 s at a 30 s interval.
- Setup failing specifically *after a restart* is often the previous session still holding the inverter's single Modbus slot. Closing the socket with `SO_LINGER=0` so the kernel sends a TCP RST rather than a FIN makes the inverter free the slot immediately instead of waiting out its own keep-alive.

enoch85, 2026-11-13:
The blocking call in this report, `client.scan` running in the event loop at startup, was fixed in 0.0.26, which runs the scan in the executor. cc @cjne

closed 2026-11-13 by enoch85