CJNE/ha-sunspec #285: only reconnecting after inverter restart

Zef-X, 2024-08-22:
## Version of the custom_component
0.0.26
SMA Tripower X 25 firmware 03.06.19.R

## Configuration

```yaml
Inverter not active on 192.168.20.103:502
Failed to connect to 192.168.20.103:502 slave id 126
```

## Describe the bug

After restarting HASS or restarting the integration, SunSpec wont connect to Inverter.
Stoping and starting Modbus Server or replugging Ethernet dosnt help.

When restarting Inverter, I get data as intended without problem.
But again, when I wanna change the config SunSpec tells me it can't connect to Inverter even though data is recieved and everything works as intended.

## Debug log

```text
2024-08-22 15:11:36.089 WARNING (MainThread) [custom_components.sunspec] Inverter not active on 192.168.20.103:502
2024-08-22 15:12:25.419 WARNING (MainThread) [custom_components.sunspec] Failed to connect to 192.168.20.103:502 slave id 126

```

SmileyMan, 2024-08-23:
Same. Reverted to 0.0.25

SmileyMan, 2024-09-16:
Temp fix for me was downgrading to 0.0.25 and replacing sensor.py with sensor.py from 0.0.26 (Constants in use by 0.0.25 get removed Jan 2025).

Suspect "Bump pysunspec2 from 1.0.8 to 1.1.5" is the source of this new issue.

borrelan, 2024-09-22:
That worked for me as well.  Thank you for pointing me in the right direction @SmileyMan

eshattow, 2024-11-09:
I see the error "Failed to connect to host (hostname):502 slave 1 - Inverter not active on (hostname):502".

lpgn, 2025-01-06:
> Temp fix for me was downgrading to 0.0.25 and replacing sensor.py with sensor.py from 0.0.26 (Constants in use by 0.0.25 get removed Jan 2025).
> 
> Suspect "Bump pysunspec2 from 1.0.8 to 1.1.5" is the source of this new issue.

that worked for me as well downgraded to 0.0.25 and uploaded sensor.py from 0.0.26 Thanks @SmileyMan

jimmyppr, 2025-06-05:
> Temp fix for me was downgrading to 0.0.25 and replacing sensor.py with sensor.py from 0.0.26 (Constants in use by 0.0.25 get removed Jan 2025).
> 
> Suspect "Bump pysunspec2 from 1.0.8 to 1.1.5" is the source of this new issue.

Wow, it works! Ty! The way I did it was before the downgrade just copied the sensor.py and put it on another folder. Then proceed with the downgrade and replace the file I the same folder. First-time users the folder you're looking is located in custom_components/sunspec/sensor.py in running HAOS on Raspberry py 4.

hilman2, 2026-08-19:
"Only reconnects after an inverter restart" usually means the integration is not the one that needs to reconnect: the inverter still believes the old session is alive.

Two things cause that, and they compound:

**The socket is closed politely.** A normal `close()` sends a FIN, and an inverter with a single Modbus TCP slot often keeps that slot reserved until its own keep-alive expires, which can be minutes. Closing with `SO_LINGER=0` makes the kernel send a TCP RST instead, and the device frees the slot immediately. That one change is what turns "wait for the inverter to reboot" into "reconnects on the next cycle".

**A failed cycle reuses the same client.** If the failure was a dead socket, every retry drives the same dead socket. It needs an explicit flag that forces the next `get_client()` to discard and rebuild, and the rebuild has to force-disconnect the old one rather than letting it be garbage collected, because pysunspec2 holds an internal reference cycle and Python does not collect it promptly.

Both are implemented in [ha-sunspec2](https://github.com/hilman2/ha-sunspec2), a fork of this integration, along with an in-cycle retry that releases the gateway lock while it waits, so a second config entry on the same gateway can use the slot during the pause.

enoch85, 2026-11-13:
Hey guys, please try the latest version. It would be great to know if this issue is resolved.

eshattow, 2026-11-13:
@enoch85 writes:
> Hey guys, please try the latest version. It would be great to know if this issue is resolved.

Quoting a forum post https://forum.outbackpower.com/viewtopic.php?p=113468&hilit=home+assistant#p113468 :
> Working versions of SunSpec add-on: 0.0.24 0.0.25 0.0.27 0.0.28
> Non-working versions of SunSpec add-on: 0.0.26

closed 2026-11-13 by enoch85