CJNE/ha-sunspec #288: Sensor disabled if read operation fails 3 times

bakadave, 2024-09-19:
SunSpec version: 0.0.26

## Configuration

```yaml
not using configuration.yaml, set up via config flows.
```

## Describe the bug

Every once in a while the `AC Lifetime Energy Produced` value of the inverter goes unavailable. Apparently it happens when there is an error reading the value and the root cause may be with my Fronius inverter. Based on a short test via modbus poll it appears that the inverter occasionally does not respond to the query for that particular register.
My problem is that this value can only be read again if I restart the whole integration. Ideally the integration would keep trying to read the value even if there is no response.

## Debug log

Relevant Home assistant log entry:
```text

2024-09-17 19:21:24.086 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: Fimer100: Error: device: 1 address: 40233 -> Modbus Error: [Input/Output] ERROR: No response received after 3 retries

```

hilman2, 2026-08-19:
Worth separating two things that look the same from the outside: an entity going unavailable because a *read* failed, and an entity going unavailable because the *whole update cycle* failed. With HA's `DataUpdateCoordinator` it is almost always the second, and three consecutive failures at a 30 s interval is a minute and a half of a flaky network, which is not much on an inverter link.

What works, from doing it in [ha-sunspec2](https://github.com/hilman2/ha-sunspec2), a fork of this integration: make `available` depend on a consecutive-failure counter rather than on `last_update_success`, and let it tolerate a few cycles so sensors keep reporting their last good value through a short outage.

One trap if anyone implements it: `DataUpdateCoordinator._async_refresh` early-returns once failures repeat, so listeners stop firing. If `available` only flips at the threshold, nothing tells HA about it and the entity stays frozen on the stale value indefinitely. `async_update_listeners()` has to be called explicitly at the moment the counter crosses the threshold.

enoch85, 2026-11-13:
@bakadave is this still an issue on the latest release?