CJNE/ha-sunspec #291: Sunny Triporwer X 12: DER DC Measurement (714) not available any more

GuentherKoll, 2024-10-07:
Not available since Update SunSpec (09.09.2024)
Some days later: DER AC Measurement (701) only a few values are available
[image]
[image]

GuentherKoll, 2024-10-08:
After reload of the integration before sunset i now get the DER AC Measurement (701).
But also after sunset i can connect to the inverter via webui (in my case https://192.168.178.92/webui/Plant:1,IGULD:SELF/dashboard)
=> in my opinion the cpu is not sleeping (see https://github.com/CJNE/ha-sunspec/issues/202)

DER DC Measurement (714) is still not available

GuentherKoll, 2024-12-03:
a disgrace for SMA (ein Armutszeugnis für SMA) ...

enoch85, 2026-11-13:
Model 714 works in 0.0.30: on Home Assistant 2026.11.1, a simulated 714 with two DC ports gave all its sensors. SMA's model list for the Sunny Tripower X supports only the per-port DC current, voltage and power of 714, and SMA says these registers read NaN while the inverter's CPU is off at night. Until 0.0.30 a sensor whose point had no value at setup was never added; since #391 it appears with the first value. cc @cjne

closed 2026-11-13 by enoch85