CJNE/ha-sunspec #294: deprecated constant "POWER_VOLT_AMPERE_REACTIVE"

MiOrt, 2024-10-08:
## Version of the custom_component

0.0.26

## Describe the bug

component uses depreciated constant

## Debug log

2024-10-08 19:26:44.534 WARNING (ImportExecutor_0) [homeassistant.const] POWER_VOLT_AMPERE_REACTIVE was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'sunspec' custom integration

```

enoch85, 2025-01-30:
@CJNE Hi, september is soon. :)

MasterAndi, 2025-06-01:
Still the Error Message in the HA Log

matze338, 2025-09-16:
Can we update Homeassistent to the 2025.9 version or we will have than errors in sunspec instead of warnings?

mp1981, 2025-09-16:
I've changed the file as mentioned by maxilampert and it works without errors or warnings before and after HA Update

StefanW86, 2025-09-18:
Changed it manually in 2 systems. Works well.

enoch85, 2026-11-12:
Fixed in 0.0.27: the integration uses `UnitOfReactivePower.VOLT_AMPERE_REACTIVE` since 3ea4a18, so this warning no longer appears. cc @cjne

closed 2026-11-12 by enoch85