CJNE/ha-sunspec #303: HASS 2024.11.0b0: Unable to import component: cannot import name 'Config' from 'homeassistant.core'

daemonenstall, 2024-10-31:
With the new Home Assistant Beta 2024.11.0b0, the integration can not be setup.

```text

Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 06:54:11 (1 Vorkommnisse)
Zuletzt protokolliert: 06:54:11

Setup failed for custom integration 'sunspec': Unable to import component: cannot import name 'Config' from 'homeassistant.core' (/usr/src/homeassistant/homeassistant/core.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1011, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1071, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/sunspec/__init__.py", line 13, in <module>
    from homeassistant.core import Config
ImportError: cannot import name 'Config' from 'homeassistant.core' (/usr/src/homeassistant/homeassistant/core.py). Did you mean: 'config'?

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1031, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1023, in async_get_component
    comp = self._get_component()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1071, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/sunspec/__init__.py", line 13, in <module>
    from homeassistant.core import Config
ImportError: cannot import name 'Config' from 'homeassistant.core' (/usr/src/homeassistant/homeassistant/core.py). Did you mean: 'config'?

```

gieljnssns, 2024-11-01:
With 2024.11.0b1 this is just a warning
```
Config was used from sunspec, this is a deprecated alias which will be removed in HA Core 2025.11. Use homeassistant.core_config.Config instead, please report it to the author of the 'sunspec' custom integration
```

enoch85, 2026-11-12:
Fixed in 0.0.27: `Config` is imported from `homeassistant.core_config` since cf6f5a2. cc @cjne

closed 2026-11-12 by enoch85