CJNE/ha-sunspec #304: Regression 0.0.26 Failed to connect, check hostname and port

eshattow, 2024-11-09:
Hello,

Core 2024.11.1
Supervisor 2024.11.2
Operating System 13.2
Frontend 20241106.2

SunSpec integration version 0.0.26 is fails SunSpec Modbus TCP connection at Add Integration action, with error:
> Failed to connect, check hostname and port

Removal and installing version 0.0.25 allows a successful SunSpec Modbus TCP connection at Add Integration action.

ulc6, 2024-11-10:
Hello,

same problem in my system.

I had "this night" a complete "power-off/power-on" of my inverter, but IP adress has not changed/is fix and I didn't noticed changes in my setting.

HA:
Core 2024.11.1
Supervisor 2024.11.2
Operating System 13.2
Frontend 20241106.2

Inverter:
SMA Tripower X
Firmware version 03.06.19.R

rednaxela14, 2024-12-30:
ID 126?

ulc6, 2024-12-31:
@rednaxela14 
Yes, ID 126 (default)

An additional Information:
Sometimes sunspec is working, but in this cases only information model 701 DER AC Measurement is working. Information model 714 DC Monitoring is not working... but it has working too in the past.

closed 2025-11-10 by eshattow