CJNE/ha-sunspec #306: self.config_entry = config_entry. This will stop working in Home Assistant 2025.12

BRutuS81, 2024-12-22:
HI, 

I did not find the exact same issue, hopefully I did not create it double. 

**1st one:**
Logger: homeassistant.helpers.frame
Bron: helpers/frame.py:324
First seen: 18:33:01 (1 gebeurtenissen)
Last logged: 18:33:01

Detected that custom integration 'sunspec' sets option flow config_entry explicitly, which is deprecated at custom_components/sunspec/config_flow.py, line 142: self.config_entry = config_entry. This will stop working in Home Assistant 2025.12, please create a bug report at https://github.com/cjne/ha-sunspec/issues

**2nd one:**
Logger: homeassistant.const
Bron: helpers/deprecation.py:222
First seen: 11:42:43 (1 gebeurtenissen)
Last logged: 11:42:43

POWER_VOLT_AMPERE_REACTIVE was used from sunspec, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'sunspec' custom integration

/homeassistant/custom_components/sunspec/config_flow.py
    def __init__(self, config_entry):                                                                        Line 140
        """Initialize HACS options flow."""                                                              Line 141
        self.config_entry = config_entry                                                                 Line 142
        self.settings = {}                                                                                          Line 143
        self.options = dict(config_entry.options)                                                    Line 144
        self.coordinator = None                                                                             Line 145

I use sunspec for my sma inverter. At this moment it is not possible to load the Energy Tab in Home Assistant due to this bug. About the 2nd bug; I already tried to rename POWER_VOLT_AMPERE_ACTIVE to VOLT_AMPERE_REACTIVE in the sensor.py file. No luck yet, but maybe also due to the 1st error (self.config_entry) which I really do not have a clue about. 
When more info is needed please let me know.

sm01-uk, 2025-12-16:
I have the same issue and having updated to Home Assistant 2025.12.3 this morning, found that the Sunspec integration now fails to load completely. Having to roll back to the previous version of HA as I currently have automations relying on the integration. Would be great if this could be fixed.

CJNE, 2026-03-15:
This may be partly outdated now.

At least some of the deprecation warnings mentioned here look like they may already have been addressed on the current branch, so this could be tied to an older release. If it still happens on the latest version, please share the SunSpec version and Home Assistant version so it can be checked properly.

enoch85, 2026-11-12:
Both warnings are fixed: `POWER_VOLT_AMPERE_REACTIVE` in 0.0.27 (3ea4a18), and since 0.0.29 the options flow no longer sets `config_entry` itself (af05fde). On Home Assistant 2026.11.1 the options open without either warning. cc @cjne

closed 2026-11-12 by enoch85