CJNE/ha-sunspec #307: Warning: deprecated alias

hvdrie, 2025-01-06:
Logger: homeassistant.core
Source: helpers/deprecation.py:222
First occurred: 10:03:14 (1 occurrences)
Last logged: 10:03:14

Config was used from sunspec, this is a deprecated alias which will be removed in HA Core 2025.11. Use homeassistant.core_config.Config instead, please report it to the author of the 'sunspec' custom integration

lechercheur123, 2025-01-07:
Hello,
You have to use HACS to download the integration: https://hacs.xyz/docs/use/
Best regards,

StefanW86, 2025-04-15:
any news here?

williwinzig, 2025-06-15:
The error message still occurs in HA 2025.6.0

StefanW86, 2025-09-18:
i updated the file manually. it works well. See pull request.

nilrog, 2025-10-15:
When can we expect a new release for this @CJNE ? The deadline for HA 2025.11 to be available is only weeks away now.