CJNE/ha-sunspec #310: Integration no longer pulling data from inverter after update to 2025.2.1

UrZdcw9, 2025-02-09:
Hi there,

Ive been a long time user of your integration.  Its served me really well to monitor consumption and solar generation - so thanks!!

However I recently updated homeassistant to 2025.2.1 and as a result the sunspec sensor are no longer showing any data.  They are all listed as unavailable. My inverter is from SolarEdge.
-->

## Version of the custom_component

I tried both v 0.25 and 0.26.
-->

## Describe the bug

Data not being pulled through to HA.

## Debug log

This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/sunspec/sensor.py:4
integration: SunSpec (documentation, issues)
First occurred: 12:31:03 (4 occurrences)
Last logged: 12:34:11

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2342, in async_forward_entry_setup
    return await self._async_forward_entry_setup(entry, domain, True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2374, in _async_forward_entry_setup
    await integration.async_get_platform(domain)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1108, in async_get_platform
    platforms = await self.async_get_platforms((platform_name,))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1185, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    platforms.update(self._load_platforms(platform_names))
                     ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1098, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1272, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1304, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/sunspec/sensor.py", line 4, in <module>
    from homeassistant.components.sensor import DEVICE_CLASS_CURRENT
ImportError: cannot import name 'DEVICE_CLASS_CURRENT' from 'homeassistant.components.sensor' (/usr/src/homeassistant/homeassistant/components/sensor/__init__.py)

rysm83, 2025-02-12:
I'm also having the same issue after updating to  2025.2

Jogi57, 2025-02-17:
Gleiches Problem, **KEINE** Daten mehr von meinen Invertern.

justme871, 2025-04-13:
Same problem here. Doesn't work anymore.

mZ738, 2025-04-13:
> Same problem here. Doesn't work anymore.

Try reinstalling.

justme871, 2025-04-13:
> > Same problem here. Doesn't work anymore.
> 
> Try reinstalling.

I can't. Are many helpers (counters) created from  sensors. I don't want to break it...

mZ738, 2025-04-13:
> > > Same problem here. Doesn't work anymore.
> > 
> > Try reinstalling.
> 
> I can't. Are many helpers (counters) created from  sensors. I don't want to break it... 

Create a Backup before. My Entities stayed the same.

I Just went into Hacs an pressed Download again. Restarted HA an it worked.

justme871, 2025-04-13:
> > > > Same problem here. Doesn't work anymore.
> > > 
> > > 
> > > Try reinstalling.
> > 
> > 
> > I can't. Are many helpers (counters) created from  sensors. I don't want to break it...
> 
> Create a Backup before. My Entities stayed the same.
> 
> I Just went into Hacs an pressed Download again. Restarted HA and it worked.

I'll try tonight. If a sensor makes a wrong report, the history to the helper is crumbled. I've had it before :(
Thanks for the advise 🍻🍻

hilman2, 2026-08-19:
For the SolarEdge users in this thread specifically, there is a failure mode that looks exactly like this and is not caused by the HA update itself.

SolarEdge inverters accept **one** Modbus TCP session at a time. If a connection attempt fails after the TCP connect succeeded but during the SunSpec scan, the already-connected client is often not torn down, and it holds that single slot. Every following attempt then genuinely finds the device busy, so the integration reports connection errors that describe its own leaked socket rather than the original fault. A Home Assistant restart is a good way to trigger the first failure, which is why the timing points at the update.

Two things to check before assuming it is a version problem:

1. Is anything else talking to the inverter? evcc, Node-RED, SolarEdge Modbus Multi, an EV charger, a Modbus proxy. Any of them holds the slot.
2. Does it recover on its own after several minutes? That points at the socket rather than at the integration, because the inverter frees the slot when its own keep-alive expires.

[ha-sunspec2](https://github.com/hilman2/ha-sunspec2), a fork of this integration, closes those leaks (v0.14.0) and closes sockets with `SO_LINGER=0` so a TCP RST makes the inverter free the slot immediately instead of waiting out its keep-alive.

enoch85, 2026-11-12:
Fixed in 0.0.26: the traceback is from the `sensor.py` of 0.0.25, which imported `DEVICE_CLASS_CURRENT`; since 69d6eca the sensors use `SensorDeviceClass`. cc @cjne

closed 2026-11-12 by enoch85