CJNE/ha-sunspec #314: SunSpec registers update each 2 intervals fails

lhac5vet, 2025-03-30:
Hi, I'm looking to understand why each 2 intervals (set to 30s) the SunSpec custom component return a timeout and no register have been read. Hereafter is the log of 2 consecutives SunSpec model fetching. First part, when fetching succeed and second part when timeout occured.  I can see that the log difference are the lines related to the "Check_Port" where the socket to my inverter is opening. It is like the socket is not open the 2nd times. 
 
``
[36m2025-03-30 13:04:53.395 DEBUG (MainThread) [custom_components.sunspec] SunSpec Update data coordinator update
[36m2025-03-30 13:04:53.395 DEBUG (MainThread) [custom_components.sunspec] Fetching models
[36m2025-03-30 13:04:53.395 DEBUG (SyncWorker_2) [custom_components.sunspec] Check_Port: opening socket on 192.168.1.150:1502 with a 3.0s timeout.
[36m2025-03-30 13:04:53.396 DEBUG (SyncWorker_2) [custom_components.sunspec] Check_Port (SUCCESS): port open on 192.168.1.150:1502
[36m2025-03-30 13:04:53.398 DEBUG (MainThread) [custom_components.sunspec] SunSpec Update data got models {160, 103}
[36m2025-03-30 13:04:53.398 DEBUG (MainThread) [custom_components.sunspec] Get data for model 160
[36m2025-03-30 13:04:53.999 DEBUG (MainThread) [custom_components.sunspec] Get data for model 103
[32m2025-03-30 13:04:54.601 INFO (MainThread) [custom_components.sunspec] Fetching sunspec data recovered
[36m2025-03-30 13:04:54.601 DEBUG (MainThread) [custom_components.sunspec] Finished fetching sunspec data in 1.206 seconds (success: True)

[36m2025-03-30 13:05:23.396 DEBUG (MainThread) [custom_components.sunspec] SunSpec Update data coordinator update
[36m2025-03-30 13:05:23.396 DEBUG (MainThread) [custom_components.sunspec] Fetching models
[36m2025-03-30 13:05:23.398 DEBUG (MainThread) [custom_components.sunspec] SunSpec Update data got models {160, 103}
[36m2025-03-30 13:05:23.398 DEBUG (MainThread) [custom_components.sunspec] Get data for model 160
[33m2025-03-30 13:05:23.999 WARNING (MainThread) [custom_components.sunspec] Response timeout
[31m2025-03-30 13:05:23.999 ERROR (MainThread) [custom_components.sunspec] Error fetching sunspec data: 
[36m2025-03-30 13:05:23.999 DEBUG (MainThread) [custom_components.sunspec] Finished fetching sunspec data in 0.603 seconds (success: False)
``
I have try to change the interval to 29s to see if any shift in time could help if there is any socket conflict but no effect.
Any idea why the socket is not open every 2 intervals ?
Thanks

hilman2, 2026-08-19:
Your observation about `check_port` is the right thread to pull. The alternating pattern is the tell.

`check_port` opens its own TCP connection to the inverter, closes it, and then the real client connects. Many inverters (SolarEdge, KACO Powador among them) accept exactly **one** Modbus TCP session at a time, so two connections in quick succession can leave the device still tearing down the first when the second arrives.

What makes it alternate rather than fail every time is the second half: when `scan()` fails after `connect()` succeeded, the client that is already connected is never torn down, because the reference is only stored on success. Nothing can reach it to close it. On a single-slot inverter that abandoned session then occupies the only slot, so the *next* attempt genuinely does find the device busy, and the one after that succeeds because the socket has finally timed out. Hence every second cycle.

Fixed in [ha-sunspec2](https://github.com/hilman2/ha-sunspec2), a fork of this integration, in v0.14.0: the client is torn down on every failure path, and the config-flow probe closes its client too, which was leaking one connected client per failed setup attempt. If you want to test the theory here without changing code, watch whether the failures stop when nothing else on your network talks to the inverter.

lhac5vet, 2026-08-27:
Hi hilman2,

Thanks for your comments and advise, I have replaced the original CJNE/ha-sunspec integration by the sunspec2 and everything  is OK excepted one point: 
My KOSTAL PICO CI 50  inverter  goes in standby when the PV power is going under a threshold and a timeout (every night at least). The TCP connection is down because its communication interface is off. 

This result in a Home assistant notification :  

[image]

How to manage this ?  could  we imagine  a automation where the SunSpec integration stop to initiate communication when a condition is no more valid or similar ?

Thanks

hilman2, 2026-08-27:
hi @lhac5vet, please open an issue in https://github.com/hilman2/ha-sunspec2

closed 2026-11-12 by enoch85