CJNE/ha-sunspec #318: Use the modern term Unit ID

sgryphon, 2025-05-30:
## Version of the custom_component

0.0.26

## Configuration

n/a

## Describe the bug

Old parameter naming for identifying the unit is used in the UI. 

The new name, in the Modbus and SunSpec documents is "Unit ID"

Steps to reproduce:

1. Go to Settings > Devices and Services > SunSpec
2. Click Configure
3. Observe in the UI the old label used for the unit identifier (the third input box)

Fix:

Update the label to use the modern term.

## Debug log

n/a

sgryphon, 2025-05-31:
I've created a PR to update this, but also had to update the devcontainer for Python 3.13 (HA has dropped support for 3.9, which the old image had by default -- although I guess you could upgrade it). I also had to update some tests for a changed constant in the latest HA (only affected tests). Once the tests were working, I made the label change.

enoch85, 2025-12-17:
I guess this can be closed?

enoch85, 2026-11-12:
Fixed in 0.0.27: the setup and options forms say Unit ID (5b99ae5), and 0.0.28 moves existing entries from `slave_id` to `unit_id` (8280151). cc @cjne

closed 2026-11-12 by enoch85